labrad/servers #228: Can't recognize scalabrad when compiling server

jwenner, 2015-10-06:
When I tried to compile the DirectEthernet server, it could not resolve the scalabrad dependency. The full traceback is below. Do I need to add anything to the PATH environment variable, or is there something else I need to do? 

```
C:\labrad-servers\servers\DirectEthernet>sbt
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256m; sup
port was removed in 8.0
Getting org.scala-sbt sbt 0.13.7 ...
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/sbt/0.
13.7/jars/sbt.jar ...
        [SUCCESSFUL ] org.scala-sbt#sbt;0.13.7!sbt.jar (2375ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/main/0
.13.7/jars/main.jar ...
        [SUCCESSFUL ] org.scala-sbt#main;0.13.7!main.jar (2124ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/compil
er-interface/0.13.7/jars/compiler-interface-bin.jar ...
        [SUCCESSFUL ] org.scala-sbt#compiler-interface;0.13.7!compiler-interface
-bin.jar (1688ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/compil
er-interface/0.13.7/jars/compiler-interface-src.jar ...
        [SUCCESSFUL ] org.scala-sbt#compiler-interface;0.13.7!compiler-interface
-src.jar (1546ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/precom
piled-2_8_2/0.13.7/jars/compiler-interface-bin.jar ...
        [SUCCESSFUL ] org.scala-sbt#precompiled-2_8_2;0.13.7!compiler-interface-
bin.jar (1639ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/precom
piled-2_9_2/0.13.7/jars/compiler-interface-bin.jar ...
        [SUCCESSFUL ] org.scala-sbt#precompiled-2_9_2;0.13.7!compiler-interface-
bin.jar (1642ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/precom
piled-2_9_3/0.13.7/jars/compiler-interface-bin.jar ...
        [SUCCESSFUL ] org.scala-sbt#precompiled-2_9_3;0.13.7!compiler-interface-
bin.jar (1624ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/action
s/0.13.7/jars/actions.jar ...
        [SUCCESSFUL ] org.scala-sbt#actions;0.13.7!actions.jar (1640ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/main-s
ettings/0.13.7/jars/main-settings.jar ...
        [SUCCESSFUL ] org.scala-sbt#main-settings;0.13.7!main-settings.jar (1658
ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/interf
ace/0.13.7/jars/interface.jar ...
        [SUCCESSFUL ] org.scala-sbt#interface;0.13.7!interface.jar (1498ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/io/0.1
3.7/jars/io.jar ...
        [SUCCESSFUL ] org.scala-sbt#io;0.13.7!io.jar (1611ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/ivy/0.
13.7/jars/ivy.jar ...
        [SUCCESSFUL ] org.scala-sbt#ivy;0.13.7!ivy.jar (1735ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/launch
er-interface/0.13.7/jars/launcher-interface.jar ...
        [SUCCESSFUL ] org.scala-sbt#launcher-interface;0.13.7!launcher-interface
.jar (1522ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/loggin
g/0.13.7/jars/logging.jar ...
        [SUCCESSFUL ] org.scala-sbt#logging;0.13.7!logging.jar (1608ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/logic/
0.13.7/jars/logic.jar ...
        [SUCCESSFUL ] org.scala-sbt#logic;0.13.7!logic.jar (1495ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/proces
s/0.13.7/jars/process.jar ...
        [SUCCESSFUL ] org.scala-sbt#process;0.13.7!process.jar (1517ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/run/0.
13.7/jars/run.jar ...
        [SUCCESSFUL ] org.scala-sbt#run;0.13.7!run.jar (1588ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/comman
d/0.13.7/jars/command.jar ...
        [SUCCESSFUL ] org.scala-sbt#command;0.13.7!command.jar (1716ms)
downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/classp
ath/0.13.7/jars/classpath.jar ...
[... 208 lines cut ...]
le) (1405ms)
[info] downloading https://repo1.maven.org/maven2/org/scala-lang/scala-reflect/2
.11.1/scala-reflect-2.11.1.jar ...
[info]  [SUCCESSFUL ] org.scala-lang#scala-reflect;2.11.1!scala-reflect.jar (762
ms)
[info] downloading https://repo1.maven.org/maven2/org/scala-lang/modules/scala-x
ml_2.11/1.0.1/scala-xml_2.11-1.0.1.jar ...
[info]  [SUCCESSFUL ] org.scala-lang.modules#scala-xml_2.11;1.0.1!scala-xml_2.11
.jar(bundle) (222ms)
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]  ::          UNRESOLVED DEPENDENCIES         ::
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]  :: org.labrad#scalabrad_2.11;0.4.1: not found
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]
[warn]  Note: Unresolved dependencies path:
[warn]          org.labrad:scalabrad_2.11:0.4.1 (C:\labrad-servers\servers\Direc
tEthernet\build.sbt#L10-14)
[warn]            +- org.labrad:directethernet_2.11:1.2.0
[trace] Stack trace suppressed: run last *:update for the full output.
[error] (*:update) sbt.ResolveException: unresolved dependency: org.labrad#scala
brad_2.11;0.4.1: not found
[error] Total time: 7 s, completed Oct 5, 2015 8:20:15 PM
>
```

DanielSank, 2015-10-06:
I don't see anything in `build.sbt` that fetches scalabrad. I wonder if there's a way to use github as a source repo in sbt.

maffoo, 2015-10-06:
In the last change to build.sbt we took out the bintray resolver. @ejeffrey and I tested this, but it's possible we made a mistake and still had the package cached locally, so didn't realize that resolver was needed to find it. We should add that resolver line back and it will fix this issue.

@DanielSank, I would prefer to use published artifacts rather than git project references in sbt, but it's something we can explore. I have used those kind of source references in the past and had problems with them; they're not very well supported in sbt, really just sort of hacked in and not used by many people. It's something we can explore, though. I have heard that this sort of project reference is something that's being worked on for sbt 1.x, so it might get better in the near future.

maffoo, 2015-10-06:
@jwenner, when you get a chance, please try out the `u/maffoo/bintray-resolver` branch, which should fix this problem. Sorry for the inconvenience.

commit 638fe94a (2015-10-06): Add bintray resolver back to build.sbt

Fixes #228
Review: @jwenner
PR #229 (2015-10-06): Add bintray resolver back to build.sbt
Fixes #228
closed 2015-10-06 by maffoo