labrad/servers #238: ghz_fpga server should identify interface by name, not number

ejeffrey, 2015-10-07:
The boardgroups registry key identifies a board group by (ethernet_server_name, interface #) where the interface # is based on the pcap enumeration order.  However, it seems that this order is not particularly stable.  The direct ethernet server 'adapters' setting returns both an index and a name.  We should use the name in the boardgroup key, that should be more stable.

jwenner, 2015-10-08:
At least for anyone using the Delphi DirectEthernet server, all NICs on the motherboard are identified with the same name (see the call to `direct_ethernet.adapters` in #232). Further, plug-in NICs can also share the same name. Hence, the name is not a unique identifier with the Delphi server.

ejeffrey, 2015-10-08:
Hmm... OK. That is a but in the direct ethernet server, it should be labeling the devices as "Local Area Connection 1" or "PCS network" not the driver name.

ejeffrey, 2015-10-08:
OK, this is a pretty easy fix to the delphi direct ethernet server, it is just using the wrong field from winpcap.  I am happy to fix this if it will make things easier.

maffoo, 2015-10-08:
The scala version of the fpga server uses the name field (which is unambiguous) instead of the description field (which is human readable, but optional and potentially ambiguous). It'd be possible to change this to, for example, combine both fields. I agree that a string name is better than a number.

ejeffrey, 2015-10-08:
Note that on Windows, the 'description' is the driver name and not changeable by the user, while the interface name is user accessible, so actually tends to be more human readable on machines with multiple interfaces (which is the only thing that matters!)

On linux, the device names are a bit less user friendly, they tend to have names like 'eth0' or 'eno0', but I don't even know if the description field is supported.  The interface name is what is always used.  In either case, the interface order can get changed when new devices (or drivers like vmware's virtual devices) get added or removed.

maffoo, 2015-10-08:
@ejeffrey, I think maybe the description and name fields are reversed, compared to what you said. The scala direct ethernet server currently uses the name field and the names it reports on windows are something like `\\Device\\NPF_{2EC9358D-FB28-40BD-889C-A563E1EDF2BC}` (see #232).

ejeffrey, 2015-10-08:
Yikes.  OK, well I didn't even know that was an option :).

The description field is definitely the thing I mentioned: the
driver/hardware name, and is fixed.  It sounds like the name you want isn't
actually an available field.

jwenner, 2015-10-21:
@ejeffrey, @maffoo, I just realized that these weird Windows adapter names can be determined from Wireshark (by going to `Wireshark>tshark.exe -D`); this provides the conversion to names such as FPGA/PCS/GPIB/Gigabit. It's possible they could come from WinPCap (NPF stuff at https://www.winpcap.org/docs/docs_412/html/group__NPF.html)