labrad/servers #239: GHz FPGAs crashes after update to direct ethernet update to v1.2.1

btchiaro, 2015-10-11:
Scans occasionally, but not always, crash after restarting the new direct ethernet server and / or GHz FPGA server.  Below is the traceback.

```
In [4]: bq.t1(s,['q1'])

---------------------------------------------------------------------------

Error                                     Traceback (most recent call last)

<ipython-input-4-d20fc77930c7> in <module>()

----> 1 bq.t1(s,['q1'])

/home/ben/pyleBen/pyle/pyle/dataking/batch_qubit.pyc in t1(Sample, measure, delay, state, stats, name, save, collect, noisy, **kwargs)

    521 

    522     data = sweeps.grid(func, axes, dataset=dataset, save=save, noisy=noisy,

--> 523                        **kwargs)

    524 

    525     if collect:

/home/ben/pyleBen/pyle/pyle/dataking/sweeps.pyc in grid(func, axes, dataset, **kw)

    320         dataset=dataset,

    321         abortPrefix=[1],

--> 322         **kw

    323     )

    324 

/home/ben/pyleBen/pyle/pyle/dataking/sweeps.pyc in _run(func, sweep, save, dataset, abortable, abortPrefix, collect, noisy, pipesize)

    219         params = dataset.params

    220         name = dataset.name

--> 221         data = np.vstack(it)

    222         kwargs = {'number': dataset.num, 'path': dataset.path} if save else {}

    223         return pyle.datavault.LocalDataSet(data, independents, dependents,

/usr/local/lib/python2.7/dist-packages/numpy/core/shape_base.pyc in vstack(tup)

    226 

    227     """

--> 228     return _nx.concatenate([atleast_2d(_m) for _m in tup], 0)

    229 

    230 def hstack(tup):

/home/ben/pyleBen/pyle/pyle/datasaver.pyc in capture(self, iterable)

    146         """

    147         with self as dataset:

--> 148             for data in iterable:

    149                 dataset.add(data)
[... 1296 lines cut ...]

at org.labrad.Reflect$$anonfun$15$$anonfun$apply$4.apply(reflect.scala:47)

at org.labrad.Server$$anonfun$org$labrad$Server$$handleRequest$1$$anonfun$apply$7.apply(Server.scala:174)

at org.labrad.Server$$anonfun$org$labrad$Server$$handleRequest$1$$anonfun$apply$7.apply(Server.scala:159)

at org.labrad.Server$.handle(Server.scala:218)

at org.labrad.Server$$anonfun$org$labrad$Server$$handleRequest$1.apply(Server.scala:159)

at org.labrad.Server$$anonfun$org$labrad$Server$$handleRequest$1.apply(Server.scala:159)

at org.labrad.util.AsyncSemaphore$$anonfun$map$1.apply(AsyncSemaphore.scala:83)

at org.labrad.util.AsyncSemaphore$$anonfun$map$1.apply(AsyncSemaphore.scala:82)

at scala.util.Success$$anonfun$map$1.apply(Try.scala:237)

at scala.util.Try$.apply(Try.scala:192)

at scala.util.Success.map(Try.scala:237)

at scala.concurrent.Future$$anonfun$map$1.apply(Future.scala:235)
```

maffoo, 2015-10-20:
The error message is in there, if rather obscured by the verbose traceback: `context: (10,14). must connect to an adapter first.` This must be an issue with the ghz_fpga_server not properly setting up its communication with the direct ethernet server after the server restarts.

maffoo, 2015-10-21:
Just realized that the problem here is almost certainly the same as #235

maffoo, 2015-10-21:
@btchiaro: closed this based on the assumption that it's the same issue as #235. If you encounter further problems, please reopen or raise a new issue.

commit a1950d3d (2015-10-21): Catch and log exceptions when removing board groups.

Also makes board group removal itself more robust by checking
whether the direct ethernet server is still connected before
trying to expire the board group context.

Fixes #235, Fixes #239
Review: @ejeffrey
closed 2015-10-21 by maffoo