labrad/servers #244: Qubit server startup script is dumb, remove or fix it

ejeffrey, 2015-10-16:
The qubit server startup script (bin/qubitserver) deliberately overrides the system version of java with whatever version is reported by javac (the java compiler), causing it to break on systems where java8 is the default, but the java7 compiler is default.

In addition, it gets into a state where it doesn't respond to kill or ctrl-C, and requires kill-9, which should never ever happen, and makes it much harder to kill and fix.

maffoo, 2015-10-16:
I haven't looked into the startup script; we're just using the default on generated by the pack plugin. However, you can just run `java -cp lib/* org.labrad.qubits.QubitServer` and bypass the startup script altogether if you want.

ejeffrey, 2015-10-16:
Yeah, just removing the startup script would be fine.

maffoo, 2015-10-16:
Also, it looks like if you invoke the startup script like `JAVACMD=java bin/qubitserver` that would work too, since it will override the bits that look at javac to get the java path.

ejeffrey, 2015-10-16:
Yeah, my concern is when people are trying to start the serve rmanally.  The node uses the command from the INI file which works fine, but that wasn't working yesterday and Chris tried to start from the script and it didn't work in whatever user environment he had.  It seems better to just make people invoke java directly than to have a script present that (to my eyes) appears to go out of its way to do the wrong thing.  It seems maybe designed around running on a developers machine rather than deployed to production.  I am also concerned about it becoming 'unkillable' when the version is wrong, that is really confusing for people who don't know how to go to another terminal and kill -9 the process.  Probably that is a bug in one of the frameworks.