labrad/servers #258: Can't start serial server

DanielSank, 2015-11-08:
When I try to run the serial server on Vince, I get the following (horribly formatted) error message. You may want to view this in edit mode.

```
[node Vince] Remote Traceback (most recent call last):
Failure: labrad.types.Error: (0) Startup failed. [payload=[(datetime.datetime(2015, 11, 8, 14, 16, 18, 465675), "WARNING:root:Twisted version >= 14.0.0 required for SSL support. Older versions don't support the platform CA roots.\n"), (datetime.datetime(2015, 11, 8, 14, 16, 18, 572125), '2015-11-08 14:16:18-0800 [-] Log opened.\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 572250), '2015-11-08 14:16:18-0800 [-] Starting factory <__main__.SerialServer instance at 0x2b3f1b8>\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 573445), '2015-11-08 14:16:18-0800 [Uninitialized] Connected to 127.0.0.1:7682\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 619965), '2015-11-08 14:16:18-0800 [ServerProtocol,client] Vince Serial Server starting...\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 963662), "2015-11-08 14:16:18-0800 [ServerProtocol,client] 'connection failed, disconnecting'\n2015-11-08 14:16:18-0800 [ServerProtocol,client] Traceback (most recent call last):\n"), (datetime.datetime(2015, 11, 8, 14, 16, 18, 963790), '2015-11-08 14:16:18-0800 [ServerProtocol,client] File "/usr/local/labrad/pylabrad/labrad/server.py", line 404, in _connectionMade\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 963840), '2015-11-08 14:16:18-0800 [ServerProtocol,client] yield self._initServer()\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 963883), '2015-11-08 14:16:18-0800 [ServerProtocol,client] File "/usr/lib64/python2.7/site-packages/twisted/internet/defer.py", line 1045, in _inlineCallbacks\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 964118), '2015-11-08 14:16:18-0800 [ServerProtocol,client] result = g.send(result)\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 964187), '2015-11-08 14:16:18-0800 [ServerProtocol,client] File "/usr/local/labrad/pylabrad/labrad/server.py", line 440, in _initServer\n2015-11-08 14:16:18-0800 [ServerProtocol,client] yield self.initServer()\n2015-11-08 14:16:18-0800 [ServerProtocol,client] File "serial_server.py", line 72, in initServer\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 964277), '2015-11-08 14:16:18-0800 [ServerProtocol,client] self.enumerate_serial_pyserial()\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 964341), '2015-11-08 14:16:18-0800 [ServerProtocol,client] File "serial_server.py", line 110, in enumerate_serial_pyserial\n2015-11-08 14:16:18-0800 [ServerProtocol,client] dev_list = serial.tools.list_ports.comports()\n2015-11-08 14:16:18-0800 [ServerProtocol,client] File "/usr/lib/python2.7/site-packages/serial/tools/list_ports_posix.py", line 122, in comports\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 964452), '2015-11-08 14:16:18-0800 [ServerProtocol,client] return [(d, describe(d), hwinfo(d)) for d in devices]\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 964514), '2015-11-08 14:16:18-0800 [ServerProtocol,client] File "/usr/lib/python2.7/site-packages/serial/tools/list_ports_posix.py", line 93, in describe\n2015-11-08 14:16:18-0800 [ServerProtocol,client] return usb_lsusb_string(sys_usb)\n2015-11-08 14:16:18-0800 [ServerProtocol,client] File "/usr/lib/python2.7/site-packages/serial/tools/list_ports_posix.py", line 80, in usb_lsusb_string\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 964562), "2015-11-08 14:16:18-0800 [ServerProtocol,client] return base\n2015-11-08 14:16:18-0800 [ServerProtocol,client] NameError: global name 'base' is not defined\n"), (datetime.datetime(2015, 11, 8, 14, 16, 18, 964961), "2015-11-08 14:16:18-0800 [ServerProtocol,client] There was an error: global name 'base' is not defined\n"), (datetime.datetime(2015, 11, 8, 14, 16, 18, 965015), '2015-11-08 14:16:18-0800 [ServerProtocol,client] Stopping factory <__main__.SerialServer instance at 0x2b3f1b8>\n'), (datetime.datetime(2015, 11, 8, 14, 16, 18, 965276), '2015-11-08 14:16:18-0800 [-] Main loop terminated.\n')]]
```

maffoo, 2015-11-08:
This looks like a bug in pyserial. The relevant section from `list_ports_posix.py`:
```python
    def usb_lsusb_string(sysfs_path):
        bus, dev = os.path.basename(os.path.realpath(sysfs_path)).split('-')
        try:
            desc = popen(['lsusb', '-v', '-s', '%s:%s' % (bus, dev)])
            # descriptions from device
            iManufacturer = re_group('iManufacturer\s+\w+ (.+)', desc)
            iProduct = re_group('iProduct\s+\w+ (.+)', desc)
            iSerial = re_group('iSerial\s+\w+ (.+)', desc) or ''
            # descriptions from kernel
            idVendor = re_group('idVendor\s+0x\w+ (.+)', desc)
            idProduct = re_group('idProduct\s+0x\w+ (.+)', desc)
            # create descriptions. prefer text from device, fall back to the others
            return '%s %s %s' % (iManufacturer or idVendor, iProduct or idProduct, iSerial)
        except IOError:
            return base

    def describe(device):
        """\
        Get a human readable description.
        For USB-Serial devices try to run lsusb to get a human readable description.
        For USB-CDC devices read the description from sysfs.
        """
        base = os.path.basename(device)
        # USB-Serial devices
        sys_dev_path = '/sys/class/tty/%s/device/driver/%s' % (base, base)
        if os.path.exists(sys_dev_path):
            sys_usb = os.path.dirname(os.path.dirname(os.path.realpath(sys_dev_path)))
            return usb_lsusb_string(sys_usb)
        # USB-CDC devices
        sys_dev_path = '/sys/class/tty/%s/device/interface' % (base,)
        if os.path.exists(sys_dev_path):
            return read_line(sys_dev_path)
        return base
```
The error is coming from the first `return base`.

This is not the most recent version of pyserial; can you just upgrade?

pomalley, 2015-11-08:
> This is not the most recent version of pyserial; can you just upgrade?

Probably not, unfortunately. @zchen088 can you weigh in here on the different versions of pyserial?

maffoo, 2015-11-08:
This can be fixed simply by just passing in the `base` argument from `describe` into `usb_lsusb_string`:
```python
    def usb_lsusb_string(sysfs_path, base):
        ...

    def describe(device):
        ...
            return usb_lsusb_string(sys_usb, base)
        ...
```

zchen088, 2015-11-08:
We've traditionally had problems with the most recent version of ```pyserial``` giving us mysterious timeouts, though nobody knows why.