labrad/servers #269: DAC Bringup script in main servers directory

jwenner, 2015-11-19:
In #226, I thought the plan was to put `GHz_DAC_bringup` in a scripts (or fpgascripts) folder. However, when it got merged, it wound up in the top directory with all the server files. To me at least, this makes it harder to distinguish that it is a script instead of a server. Should it go into a scripts or fpgascripts folder, back to GHzDACs, or be left as is? @maffoo, @DanielSank

DanielSank, 2015-11-19:
I definitely don't like it being in the top level. I think we should

1. Move all of the servers into a subdirectory of the git repo root directory.
2. Put *everything* related to the FPGAs in a sub-package.

jwenner, 2016-01-25:
@maffoo, @DanielSank, after discussing with John I'll be moving the DACScanPhases and DACPrintPhases scripts into this repository from pyle. However, I don't believe these should clutter up the top layer with all the servers. So, what do we want to do with these scripts plus the bringup script (since all three are scripts specific to the DACs)?