labrad/servers #271: mks gauges server incorrectly reading gauges

zchen088, 2015-11-22:
On Vince, they all read 1:
```
In [4]: cxn.mks_gauge_server.get_named_readings()                               
Out[4]: [('Still', Value(10.0, 'Torr')), ('Pot Low', Value(10.0, 'Torr')), ('Pot High', Value(1000.0, 'Torr')), 
('Return', Value(1000.0, 'Torr')), ('Keg 1', Value(1000.0, 'Torr')), ('He Flow', Value(1.0, 'Torr'))]                           
```
I believe this has been a persistent problem on Ivan (@jwenner can you confirm)?

pomalley, 2015-11-22:
1. Are the gauges out of range? IIRC they will output 1 in this case.
2. (1) might also happen if the units get off. I don't think it would work this way but maybe the COM ports got switched and the units messed up?
3. Are the serial communications working? I think sometimes you can get an "off by one" sort of error with serial where each response has a character from the previous one, which would throw an exception when you try to `float(resp)` and then the server might give you 1.
4. Uh, I'm out of ideas.

jwenner, 2015-11-23:
@zchen088, this *had* been a problem w/ Ivan. However, on branch u/jwenner/mks_error_print, I no longer have this issue. The key for us was to change the pressure reading time, as I think the time was so short that it was sometimes reading part of the readings in the next time slot. I should probably take the time to fix #255 for this branch.