labrad/servers #273: Does the anritsu server actually query for the frequency?

zchen088, 2015-11-24:
When we do for example:
```
cxn.anritsu_server.frequency()
```
Does it query the anritsu or return some cached value?

maffoo, 2015-11-24:
Looks like it's caching. The setting reads the `.frequency` attribute (see [here](https://github.com/martinisgroup/servers/blob/master/Anritsu/anritsu_68367C.py#L89)) which gets set when the server first connects to a particular device and then gets updated when the frequency is explicitly set.

zchen088, 2015-11-24:
I see, is this intentional? It looks like the anritsu server has a ```getFrequency``` method. Why not just call that?

This was super confusing when we were trying to figure out whether the anritsu was actually responding to commands. It turns out it wasn't, but I thought it was based on asking for the frequency from the server.

ejeffrey, 2015-11-24:
This was intentional, but was a bad idea and should be fixed.  The idea was
that GPIB is slow, so we wanted to cache the value, but its better to do
that at a higher level (in the FPGA server) than to have the server itself
hold onto potentially wrong data.  frequency() should query the frequency
explicitly.

zchen088, 2015-11-24:
Ok, thanks for the clarification.

DanielSank, 2015-11-24:
+1 That caching is stupid.