labrad/servers #276: FPGA server cannot find ADCs or DACs

Coxious, 2015-12-01:
Hi,

I'm a student currently studying in Haohua's group, Zhejiang University, China. Recently we want to upgrade our LabRad system to latest version on GitHub, and we meet some trouble here.

We deployed the latest LabRad system, and it cannot find our ADCs and DACs. Calling the list device function can properly print those boards on screen. However when I select an ADC or DAC boards, it shows that it's not a DAC or ADC board. 

We cloned the entire server project and pylabrad project from github.

https://github.com/martinisgroup/servers
https://github.com/maffoo/pylabrad

We run the Delphi version of Direct Ethernet ,1.1.1.

 https://github.com/martinisgroup/servers/blob/master/ghz_fpga_server.py : 1062
```python
    # Allow selecting different kinds of devices in each context.
    def selectedDAC(self, context):
        dev = self.selectedDevice(context)
        if not isinstance(dev, dac.DAC):
            raise Exception('selected device is not a DAC board')
        return dev
```
 I printed the results of 
`dev`, and it's a `DeviceWrapper` instance rather than `DAC` instance. Although `DAC` is a subclass of `DeviceWrapper`, but I looked around and only found that the instance is constructed like this:

https://github.com/maffoo/pylabrad/blob/master/labrad/devices.py : 101

```python

class DeviceServer(LabradServer):
    """A server for devices.
    
    Creates a DeviceWrapper for each device it finds, based on a
    user-provided function.  Provides standard settings for listing
    devices, selecting a device for the current context, and
    refreshing the list of devices.  Also, provides for device-locking
    with timeouts.
    """
    name = 'Generic Device Server'
    deviceWrapper = DeviceWrapper
```

and 

https://github.com/maffoo/pylabrad/blob/master/labrad/devices.py : 163

```python
        # start additions
        for name, args, kw in additions:
            if name in self.device_guids:
                # we've seen this device before
                # so we'll reuse the old guid
                guid = self.device_guids[name]
            else:
                guid = self.device_guids[name] = self._next_guid
                self._next_guid += 1
                
            dev = self.deviceWrapper(guid, name)
            yield dev.connect(*args, **kw)
            self.devices[guid, name] = dev
```
Where the `dev` was constructed.

I didn't find where or how did the devices initialized with class ADC or DAC.  

Can you guys help me ? Thanks a lot.

DanielSank, 2015-12-01:
> We cloned the entire server project and pylabrad project from github.
>
> https://github.com/martinisgroup/servers
> https://github.com/maffoo/pylabrad

Let's start by using the right repositories. `github.com/maffoo/pylabrad` is not official. Please use `github.com/labrad/pylabrad`.

Coxious, 2015-12-01:
Ah.. I see. I'll try that.

DanielSank, 2015-12-01:
Whenever you post an issue, please give your *exact* python shell inputs and outputs. This is much more important than posting the server code, which we already have. Remember, your goal is to help us understand *your* problem, so we need to see *your* code.

Coxious, 2015-12-01:
We didn't actually write much code this time, we just deployed the code and run the code. I'll change the pylabrad version and see if there is any problem and supplement more information if needed. Thanks again.

DanielSank, 2015-12-01:
> I printed the results of dev, and it's a DeviceWrapper instance rather than DAC instance.

I'd like to see *exactly* what you did here.

Coxious, 2015-12-01:
Oh, I see. Sorry for that. I thought I proper described it. I'm not in the lab now and please wait a few more minutes for me to get the picture...

Coxious, 2015-12-01:
I just checked our console prints.

https://github.com/martinisgroup/servers/blob/master/ghz_fpga_server.py : 1062

```
    # Allow selecting different kinds of devices in each context.
    def selectedDAC(self, context):
        dev = self.selectedDevice(context)
        print dev # I added this
        if not isinstance(dev, dac.DAC):
            raise Exception('selected device is not a DAC board')
        return dev
```

Here is the result:
[image]
[image]

gribeill, 2015-12-04:
Hi all,

We're seeing what I think is a very similar issue at UW-Madison when trying to run the latest `ghz_fpga_server.py` (version 5.2.0). This is on a machine with a fresh install of labrad, including:
`pylabrad-0.93.1`
Scala LabRAD manager `scalabrad-0.5.1`
Scala `DirectEthernet-1.2.2`
The PC is connected to a single DAC board (build 8) over Ethernet.

When running `ghz_fpga_server.py`, we see the following output (note the timeout exceptions):
```
2015-12-03 18:59:47-0600 [ServerProtocol,client] GHz FPGAs starting...
2015-12-03 18:59:47-0600 [-] Refreshing client connection...
2015-12-03 18:59:47-0600 [-] Loading board group definitions from registry...
2015-12-03 18:59:47-0600 [-] Board group definitions loaded.
2015-12-03 18:59:47-0600 [-] Board group definitions ok.
2015-12-03 18:59:47-0600 [-] Adapter "Leiden Direct Ethernet" (port 1) does not exist. Group will not be added.
2015-12-03 18:59:47-0600 [-] Adapter "Shasta Direct Ethernet" (port 1) does not exist. Group will not be added.
2015-12-03 18:59:47-0600 [-] Board groups to be added: set([('mcdermott5125-2 Direct Ethernet', 2L)])
2015-12-03 18:59:47-0600 [-] Board groups to be removed: set([])
2015-12-03 18:59:47-0600 [-] Creating board group "mcdermott5125-2 Board": server="mcdermott5125-2 Direct Ethernet", port=2
2015-12-03 18:59:47-0600 [-] {('mcdermott5125-2 Direct Ethernet', 2L): <__main__.BoardGroup object at 0x043330B0>}
ERROR:root:timeout exception: (12) Timeout! Failed to collect 1 packets after 1 second [payload=None]
ERROR:root:timeout exception: (12) Timeout! Failed to collect 1 packets after 1 second [payload=None]
2015-12-03 18:59:49-0600 [-] Devices detected on board group "mcdermott5125-2 Board":
2015-12-03 18:59:49-0600 [-]   mcdermott5125-2 Board DAC 11
2015-12-03 18:59:49-0600 [ServerProtocol,client] GHz FPGAs now serving
```
We see the same timeout error that @Coxious sees when running the Delphi DirectEthernet server. Reading through old issues, I feel like this is similar to #232, although that should be a closed issue so I don't understand what in our configuration could be causing this.

Some further data points:
1. I can access the ghz_fpgas servers with pylabrad, and it will list the correct boards and dacs, but dac-specific functions will fail:
```
>>> import labrad
>>> cxn = labrad.connect()
Enter LabRAD password:
>>> fpgas = cxn.ghz_fpgas()
>>> fpgas.list_devices()
[(0L, 'mcdermott5125-2 Board DAC 11')]
>>> fpgas.list_dacs()
['mcdermott5125-2 Board DAC 11']
>>> fpgas.select_device('mcdermott5125-2 Board DAC 11')
'mcdermott5125-2 Board DAC 11'
>>> fpgas.dac_leds(10)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\labrad\client.py", line 63, in __call__
    return future.wait() if wait else future
  File "C:\Python27\lib\site-packages\labrad\backend.py", line 383, in wait
    result.raiseException()
  File "<string>", line 2, in raiseException
labrad.types.Error: (0) [GHz FPGAs] Remote Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1128, in _inlineCallbacks
    result = g.send(result)
  File "C:\Users\McDermott Lab\Desktop\servers-qubit-server-v0.7.0\ghz_fpga_server.py", line 1750, in dac_leds
    dev = self.selectedDAC(c)
  File "C:\Users\McDermott Lab\Desktop\servers-qubit-server-v0.7.0\ghz_fpga_server.py", line 1035, in selectedDAC
    raise Exception('selected device is not a DAC board')
exceptions.Exception: selected device is not a DAC board
 [payload=None]
>>> fpgas.build_number()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\labrad\client.py", line 63, in __call__
    return future.wait() if wait else future
  File "C:\Python27\lib\site-packages\labrad\backend.py", line 383, in wait
    result.raiseException()
  File "<string>", line 2, in raiseException
labrad.types.Error: (0) [GHz FPGAs] Remote Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1128, in _inlineCallbacks
    result = g.send(result)
  File "C:\Users\McDermott Lab\Desktop\servers-qubit-server-v0.7.0\ghz_fpga_server.py", line 1594, in build_number
    buildNumber = yield dev.buildNumber()
exceptions.AttributeError: DeviceWrapper instance has no attribute 'buildNumber'
 [payload=None]
```

Your thoughts on this issue are appreciated!
- Guilhem (and Alex)

DanielSank, 2015-12-04:
Can you guys give your git SHA's for each repository?

gribeill, 2015-12-04:
Assuming I'm interpreting your request correctly, since my git-fu is weak...
We forked the servers repository and built DirectEthernet from `cc7184c09e7012360965308aebd5544025ba6fbc`, and did not modify any code. scalabrad was from the bintray download.

gribeill, 2015-12-04:
One other thing I noticed: when running `ghz_fpga_server` with the Delphi DirectEthernet server, I can see the correct MAC address flash for a second during before it disappears. If you think it would help, I'm happy to try and get wireshark up and running here to take a closer look at what's going on.

maffoo, 2015-12-04:
If you want to try wireshark with the shiny new [plugin](https://github.com/martinisgroup/servers/blob/master/fpgalib/tools/ghz_fpgas.lua) that would be awesome.

gribeill, 2015-12-04:
Okay, here's what some fiddling with wireshark has revealed:

Running `ghz_fpgas_server.py`, version `5.2.0` correctly broadcasts the packets looking for GHz Boards with addresses 0-255. We are using a DAC with an address set to 11, so we see:
```
Frame 17: 70 bytes on wire (560 bits), 70 bytes captured (560 bits) on interface 0
GHz DAC
    Destination: GeocastN_aa:00:0b (00:01:ca:aa:00:0b)
    Source: 00:00:00_00:00:00 (00:00:00:00:00:00)
    Message Type: Register Write (56)
GHz DAC Register Write
    Start: no start (0)
    Readback: readback after 2 us (1)

Frame 18: 84 bytes on wire (672 bits), 84 bytes captured (672 bits) on interface 0
GHz DAC
    Destination: 00:00:00_00:00:00 (00:00:00:00:00:00)
    Source: GeocastN_aa:00:0b (00:01:ca:aa:00:0b)
    Message Type: Register Read (70)
GHz DAC Register Read
```

All this looks normal, I think, with the exception that the ethernet server seems to think its MAC is ```00:00:00:00:00:00```.

Interestingly, when I run the Delphi manager, the Delphi Ethernet Server, and `ghz_fpgas_server.py`, version `4.0`, our last stable config, we the windows' machine MAC address is set correctly:
```
Frame 16: 84 bytes on wire (672 bits), 84 bytes captured (672 bits) on interface 0
GHz DAC
    Destination: Dell_da:f3:b8 (34:17:eb:da:f3:b8)
    Source: GeocastN_aa:00:0b (00:01:ca:aa:00:0b)
    Message Type: Register Read (70)
GHz DAC Register Read
```

Is the MAC being set to all 0's an issue? To me it looks like the direct ethernet server is somehow not dealing with the inbound packets correctly, causing a timeout [here](https://github.com/martinisgroup/servers/blob/master/ghz_fpga_server.py#L412)

gribeill, 2015-12-04:
Sounds like this might be related to #268, as well.

maffoo, 2015-12-04:
The source MAC address not being set is unfortunate, but not actually a problem. We see the same behavior with the scala direct ethernet server, due to pcap not finding MAC addresses on windows, but autodetection works fine.

I think the problem is that you're using an outdated version of pylabrad. That last line of output from your python session gives a clue:
```
exceptions.AttributeError: DeviceWrapper instance has no attribute 'buildNumber'
```
That means the ghz_fpga_server is creating an object of type `DeviceWrapper` for the DAC boards it discovered, but it should be creating a more specific wrapper type based on the fpga software build, by calling the `chooseDeviceWrapper` function. If you installed pylabrad 0.93.1 from the pypi installer, then unfortunately you're way out of date because we have not really been releasing versions, just pushing to github (see labrad/pylabrad#129). Could you check out the pylabrad github repo and add it to your pythonpath, then try again? You'll want to uninstall the pylabrad package itself first.

DanielSank, 2015-12-04:
@maffoo I think we should delete the PyPI repo. What do you think?

maffoo, 2015-12-04:
I think we can add a note to PyPI indicating that it is out of date, but I think we should keep it around and start publishing updated versions. It's the canonical way to distribute python packages, and I think we should try to be good citizens of the python ecosystem.

gribeill, 2015-12-04:
OK, building from the latest pylabrad source solved it. The server can now communicate with the boards. Still getting two timeout errors: 
```
ERROR:root:timeout exception: (12) Timeout! Failed to collect 1 packets after 1 second [payload=None]
ERROR:root:timeout exception: (12) Timeout! Failed to collect 1 packets after 1 second [payload=None]
```
during board detection when we run the server, but everything seems to work.

maffoo, 2015-12-04:
Those timeouts are expected. The autodetection works by sending ping packets to every possible MAC address for DAC and ADC boards, respectively, then listening for responses until no more arrive within one second, at which point it assumes every available board has responded. So there's one timeout for DACs and one for the ADCs. We should change the log message or just eliminate it, since this is not actually an error.

maffoo, 2015-12-04:
Another thing we could do is to add a setting to the direct ethernet server to capture all packets that arrive within a certain interval and return them. This command would then return normally as opposed to the errors we get now.

Coxious, 2015-12-06:
Thanks for your discussing. After we use latest version of pylabrad everything works fine.

DanielSank, 2015-12-06:
Glad this was easy to fix. I guess we need to keep PyPI updated.

maffoo, 2015-12-08:
Closing because this seems to work for folks after upgrading pylabrad.

closed 2015-12-08 by maffoo