labrad/servers #277: Don't set master mode when doing a register readback

maffoo, 2015-12-03:
As a workaround for a hardware issue (floating daisy-in port can cause the first board in the daisy chain to emit random daisy chain pulses: GIGO), the `regPing` packet for DAC build 15 currently sets the board to master mode so that it will ignore daisy-in (see [here](https://github.com/martinisgroup/servers/blob/master/fpgalib/dac.py#L1309)). We're currently working on a hardware fix for this, namely a proper termination on daisy-in. Once that fix is in place, we should remove the master mode hack on register readback.

@DanielSank

DanielSank, 2015-12-03:
Yup, working on hardware fix. Turns out we have no resistors...

pomalley, 2015-12-07:
I understand that we need to fix the case of a floating daisy-in, but I don't think we'd actually gain anything. Other than this issue with putting floating daisy-in's in pass-through mode, it doesn't matter what mode you actually put a board in for a register readback, since every time you start a sequence you send a register packet to every board.

I guess I'm just not sure why the current state counts as a hack.

DanielSank, 2015-12-07:
>  it doesn't matter what mode you actually put a board in for a register readback

Yes, it does. When we ping the boards for e.g. timeout recovery if we set it to master start then it emits a daisy chain pulse. That can make subsequent boards fire in the middle of the recovery procedure. At best this increments their execution counters by an unknown amount. At worst it does something else.

ejeffrey, 2015-12-07:
Also, setting the board to master (and probably slave) resets the execution
counter to zero, defeating the purpose of the register readback.