labrad/servers #281: Jump Table Commands in Wrong Order

andrewdunsworth, 2015-12-03:
Once like every hundred scans or so the jump table command will happen before the pulse sequence (or visa versa).  I noticed this in my version of spin echo where the second pi/2 pulse would sometimes just jump right next to the pi pulse, but I think this happens during our version of swap spectroscopy as well as there will be random points at long delay times that have a very large population (as if the pi pulse is jumping right next to the readout pulse).  Can someone packet sniff this out? @maffoo @DanielSank @zchen088 @JulianSKelly

zchen088, 2015-12-03:
Pictorial example:
We want this:
pi/2 ----- pi ------ pi/2 measure
But we get this:
pi/2 ----- pi pi/2 ------ measure

andrewdunsworth, 2015-12-03:
This was in the SiQ branch under s = ['','SiQ','data', 'Qubit', 'SQSi Pla 16', 'Dry', '151203-Cal3'] to be specific (sq.spin_echo_long).

DanielSank, 2015-12-03:
Looks like a delay and a pi/2 pulse are switching places?

andrewdunsworth, 2015-12-03:
yes.

maffoo, 2015-12-03:
I don't think this is a problem for packet sniffing, at least not yet. The first thing you should do is look at the jump table commands your code is actually generating to send to the servers. If you rerun your code, do you get blips at the same location? (That is, do the same inputs to your scan function reliably generate bad sequences?)

zchen088, 2015-12-05:
We ran our scan with the same input multiple times and it only occasionally failed. Also, we noticed that within the same run of the experiment, some of the stats would be bad and some of the stats would be good.

Here's a good run:
[image]

which corresponds to:
```Play SRAM - IDLE - Play SRAM - IDLE - Play SRAM```

and here's a bad run with the exact same parameters
[image]

which looks more like:
```Play SRAM - Short IDLE - Play SRAM - Play SRAM - ????```

Does this point to more of a hardware and firmware issue?

JulianSKelly, 2015-12-05:
Given that the software will always be outputting the same thing I think this indicates a hardware failure.

maffoo, 2015-12-05:
within the same run of the experiment, some of the stats would be bad and some of the stats would be good.

Does this mean that after a single start command, some of the N repetitions output a correct sequence and some an incorrect sequence? That would _definitely_ be an fpga hardware problem.

zchen088, 2015-12-05:
Yes it appears so :(

ejeffrey, 2015-12-05:
Could also be SRAM writes happening in the middle of the sequence.

pomalley, 2015-12-05:
Yes, we've definitely had issues with SRAM writes happening while the
sequence is running, at least when debugging. I don't know that it has been
thoroughly tested in the wild, e.g. with long scans and stuff.

To look at this, you can plug two of the scope lines into the mon1 and mon2
connectors on the back of the board, and then play around with what gets
written to the monitors (i.e. you'd want to look at SRAM start and sequence
start, etc.)

pomalley, 2015-12-05:
Specifically here is the code you need to change to change the monitors (IIRC, things have been shifted a bit recently):

https://github.com/martinisgroup/servers/blob/master/fpgalib/dac.py#L1205

zchen088, 2015-12-05:
I'll try to get started debugging, but can we get someone with more GHzDAC-fu to help us next week? This seems important.

DanielSank, 2015-12-05:
> can we get someone with more GHzDAC-fu to help us next week? This seems important.

Absolutely. I finished the data run on Vince and have nothing major on my todo list. Happy to help. Some things you guys could do:

* Peter's suggestion about looking at the monitors.

* Install Wireshark and @maffoo's plug-in for dissecting our FPGA packets.

* Modify fpgaseqtransmon to log the packets it's preparing for the qubit sequencer and make sure they're right. I think there may already be a flag you can set in the module so that it writes packet data to a file.

DanielSank, 2015-12-09:
@andrewdunsworth and/or @zchen088  can you summarize what we learned yesterday here?

andrewdunsworth, 2015-12-09:
When this problem occurs it occurs on every stat of an experiment.  In the case of sq.swap_spectroscopy it seemed to only effect one board (the z board) where instead of running the step_down, jump table idle, and step_up it ran the step_down, step_up (all that was written in SRAM) repeatedly every 16 us. i.e:

intended scan: -----|_________|---------
actual scan:     -----|_|------------|_|----------|_|----...

the readout, xy, and adc boards did not seem to be effected by this problem.

We then learned that the way the DAC boards work is that the daisy chain fires on every stat of an experiment (which wasn't obvious to me).

andrewdunsworth, 2015-12-09:
actual scan*:  ---------|_|----------|_|----------|_|-----...

DanielSank, 2015-12-09:
@andrewdunsworth if you put backticks around some text it shows up as a fixed with font. You can also fence a block of code with three back ticks.

andrewdunsworth, 2015-12-09:
ok it's getting rid of my underscores in the middle but you get the idea.

maffoo, 2015-12-09:
Ok, so problem occurs on every repetition of a given experiment. Does it occur every time that particular sequence is run?

DanielSank, 2015-12-09:
> Ok, so problem occurs on every repetition of a given experiment.

We are not 100% sure of that.

> Does it occur every time that particular sequence is run?

No.

andrewdunsworth, 2015-12-09:
No, it happens seemingly randomly about every 10 to 100 times a given scan is run.

andrewdunsworth, 2015-12-09:
And only if jump table commands are called (we are using idle, nop, and end).

DanielSank, 2015-12-09:
@andrewdunsworth you can edit your comment posts after they're posted, fyi.

andrewdunsworth, 2015-12-09:
>No, it happens seemingly randomly about every 10 to 100 times a given scan is run.

andrewdunsworth, 2015-12-09:
Here are the two sequences as seen on the scope:
[image]
[image]

Green is the z-board, blue is the xy, yellow is the ADC demod, and pink is the monitor_0 of the readout dac board which is set to 5 which according to 

https://github.com/martinisgroup/servers/blob/u/maffoo/dac-docs/fpgalib/docs/GHzDAC_v8.md

is the SRAM running:

[image]

the pulse on the dac monitor seems to happen before and after the sequence when the pulses occur normally, but are no where to be found when the pulses occur incorrectly.

andrewdunsworth, 2015-12-09:
When we put the adc_1 monitor on the SRAM write it seems like it writes at the same place for both the 'good' pulse sequence and the 'messed up' one:

good:
[image]

messed:
[image]

andrewdunsworth, 2015-12-09:
We put blue on monitor_0 = 5 which is the SRAM running option. It seems to know that it is running SRAM the entire time as the pulses look like this:

Good:
[image]
Bad:
[image]

maffoo, 2015-12-09:
@andrewdunsworth, do you have packet wireshark packet captures for two runs like these? It would be really helpful to see exactly what got sent to the boards.

andrewdunsworth, 2015-12-09:
Ok so if we set blue to IDLE and pink to END (I'm guessing these are the jump table commands), it looks like:
good:
[image]
bad:
[image]

So it seems to think that it is idling for the entire length of the messed up pulse sequence... which doesn't make sense (notice that there is a slight down tick where the actual idle occurs at the end of the messed up sequence).

andrewdunsworth, 2015-12-09:
This really makes it look like its a jumptable problem and not a hardware problem.

andrewdunsworth, 2015-12-09:
We are working on getting wireshark up and running.

andrewdunsworth, 2015-12-09:
We now have wireshark but are unsure how to parse the data.

andrewdunsworth, 2015-12-09:
We are now sharking wires.