labrad/servers #284: Help understanding ADC trigger tables

gribeill, 2015-12-08:
I have a few questions about the trigger table specification to make sure I'm understanding the docs correctly -- so if any of the following is way off-base, it would be great if you would let me know:

My understanding of how all this works is that the ADC sequentially steps through the trigger table when it receives the daisychain start command, up to a max of 128 entries.
Each entry is defined as a tuple (rcount, rdelay, rlength, rchan):

rchan specifies the number of channels we want to read out during this trigger (really, read out channels 0-rchan-1)

rcount specifies the number of retriggers we want to execute during this trigger table entry

on each retrigger, the ADC will wait rdelay-4 cycles, then demodulate for rlength+1 cycles using the appropriate entry in the mixer tables

So for a trigger table that looks like [(2, 100, 256, 12), (...)], the sequence would look like:

[daisychain start][(trig 1)(......delay 100 cycles....)(....demod 256 cycles.....)(......delay 100 cycles....)(....demod 256 cycles.....)][(trig 2)(...)][...]

Is this at all close to being correct? Also, given that rlength is an 8 bit number and the mixer table has 512 entries, does this mean that the maximum length of demodulation is now 1024 ns?

Thanks,
Guilhem

DanielSank, 2015-12-08:
1. Correct.

1. Correct.

1. Not sure, need to check documentation.

1. I'm not sure what `(trig1)` and `(trig2)` mean here.

> does this mean that the maximum length of demodulation is now 1024 ns?

Yes, and this can be annoying. You can, of course, fire multiple demodulations and sum the resulting IQ points to get the same SNR as you would with a longer continuous demodulation (and of course there's no reason to turn off the probe signal in between ADC triggers).

gribeill, 2015-12-08:
By (trig1), (trig 2), I meant trigger table entries.

Coxious, 2015-12-12:
@DanielSank 

I've read the document of ADC board, and I'm confused with the `rdelay`.

Here is ` Ethernet Commands for GHzADC` which you wrote in 2010.

> after performing the functions of each table entry, until it reaches a table entry with rdelay[15..8]=0, at which time the retriggering stops. Note that an empty table with rdelay[15..9]=0 at the address 0 will not ever trigger the AD.  In the table entry, rdelay[15..0]+ 3 is the number of 4 ns cycles between the AD start (or last end of ADon) and the AD trigger. 

I don't understand why we reuse the `rdelay` signal. Is this means that we have to set non-zero value to rdelay[15..8] to make sure it triggers? Then we cannot set a value less than 256, unlike @gribeill did here. This means we have to wait at least 1us until perform next trigger entry. Is that correct? 

Thanks.

ejeffrey, 2015-12-12:
@Coxious 

No, you don't have to wait 1 us.  I think what you are reading is a typo.  It should say that an entry of rdelay[15..0]=0 causes the retriggering to stop.  Since the actual delay is 'rdelay-4', zero is not otherwise valid, so you don't loose any generality.

Coxious, 2015-12-12:
Ah.. I see.. Thanks.

Coxious, 2015-12-29:
Do we have more detailed documentation for ADC trigger tables? Where can I get that? Thanks!

closed 2017-03-05 by gribeill