labrad/servers #286: Some GPIB servers do not properly report their existence when the GPIB device manager starts

DanielSank, 2015-12-08:
If we have the sampling scope server running, start the GPIB device manager, and then call `gpib_device_manager.dump_info()` we see that the sampling scope server is *not* in the list of servers known by the GPIB device manager.

However, if we have the GPIB device manager running and then start the sampling scope server, then the sampling scope server does appear in the result of the GPIB device manager's `dump_info`.

Ditto for the Hittite server.

rb25, 2015-12-09:
@andrewdunsworth and I wheeled over the sampling scope to the Jules setup, and it worked there. Jules setup uses the same scalabrad versions.

rb25, 2015-12-09:
Turns out it was related to openssl not being installed on redheatdos. It now works.

DanielSank, 2015-12-09:
> Turns out it was related to openssl not being installed on redheatdos. It now works.

The python lib? If so, please update appropriate documentation or file a ticket indicating saying that we need to indicate how to avoid this problem.

rb25, 2015-12-10:
The servers/labrad were complaining. We just didn't see it at first because there's no command line output when you start the servers through the web interface

DanielSank, 2015-12-10:
That doesn't answer my question :-)

You said

> Turns out it was related to openssl not being installed on redheatdos.

Are you talking about the python openssl package or something else, e.g. an operating system package? We need to know this so we can take the necessary actions to ensure nobody is confused by this again later.

rb25, 2015-12-10:
When starting the sampling scope server, it was warning that the python openssl package was missing.

maffoo, 2015-12-11:
Do I understand correctly that the server failed to even connect to labrad because of the missing pyOpenSSL package? If that is the case, then it's no surprise that the GPIB device manager did not know about a server that wasn't connected. The real problem would be that the server startup failure was not reported well in the web interface. Which web interface were you using?