labrad/servers #287: Qubit Sequencer is using 1.5 GB of RAM

andrewdunsworth, 2015-12-09:
Windows 7, qubit sequencer version 0.7.0, scalabrad version 0.5.1.

andrewdunsworth, 2015-12-09:
Jimmy thinks that this might be related to #281, the slowing down of mr freeze could lead to a slow down or mess up of packets being sent to the DACs. @pomalley said that he has seen the 'playing SRAM every 16 us' behavior before and thinks that it was associated with trying to write packets to the GHz DACs as they were playing.  He didn't think that this would be possible if you run experiments individually, which is what we were doing yesterday, the data rate also seems to run slower and slower the longer the qubit sequencer is running... @maffoo @DanielSank

maffoo, 2015-12-09:
If the DACs are being written to while a sequence is in progress that would be really obvious in wireshark. Have you observed this?

DanielSank, 2015-12-09:
> If the DACs are being written to while a sequence is in progress that would be really obvious in wireshark. Have you observed this?

We didn't get that far yet.

andrewdunsworth, 2015-12-09:
How do I use wire shark? And how do I interpret the data?

maffoo, 2015-12-09:
1. Download wireshark from [the website](https://www.wireshark.org/#download) and install on the machine with the NICs
2. Get the plugin from [this repo](https://github.com/martinisgroup/servers/blob/master/fpgalib/tools/ghz_fpgas.lua) and drop that file into the appropriate place for your system (the plugin docs have some hints)
3. Launch wireshark and start a capture on the NIC connected to the DAC/ADC boards. If all goes well, when you run scans you should see the captured getting recognized as DAC/ADC packets.