labrad/servers #288: Accept parameters of new qubit server doesn't compatible with old version

Coxious, 2015-12-10:
Hi,

I successfully build the latest scala version of QubitServer and struggled for a few days for a flatten error. Finally I found that the function `new_sram_block` accepts `s,w,(ss)` and  the old version of qubitserver accepts only `s,w`, so our code is no longer compatible with the new QubitServer . What does the added parameter mean ? Do you have any document about it ?  Or may I have some demo code to understand how to control the new QubitServer ?

There is another question that we are using the old version of FPGA board, DAC build 8 and ADC build 1. Is the qubit server compatible with our board ?

Here is our old version that only accepts "s,w".

```python
In [58]: cxn.qubit_sequencer.new_sram_block?
Type:           SettingWrapper
Base Class:     <class 'labrad.client.SettingWrapper'>
String Form:
LabRAD Setting: "Qubit Sequencer" >> "New SRAM Block" (ID=400)

            Create a new SRAM block with the  <...> a will be padded at the begi
nning after deconvolution.

            Accepts:
            s, w: s, w

            Returns:
```

Thanks

DanielSank, 2015-12-10:
+1

The documentation is vague.

Coxious, 2015-12-10:
We are currently using the old pyle that doesn't compatible with latest qubit server. Can we have a look at the pyle project you currently using ?

DanielSank, 2015-12-10:
> We are currently using the old pyle that doesn't compatible with latest qubit server. Can we have a look at the pyle project you currently using?

That code is not public. Let's focus on improving the documentation for the qubit sequencer so that you can update your python code appropriately.

Coxious, 2015-12-10:
Okay, thanks anyway.

Coxious, 2015-12-10:
There is another question, we are using the old version of FPGA board, DAC build 8 and ADC build 1. Is the scala QubitServer compatible with the old board ?

DanielSank, 2015-12-10:
I think so. @pomalley?

Coxious, 2015-12-10:
Oh that is great. Then maybe I can still change my code to adapt the QubitServer. Thanks !

pomalley, 2015-12-10:
It is supposed to be compatible, yes, but I don't think it's been tested since we tried it out once about nine or ten months ago.

maffoo, 2015-12-10:
Previously, the "New SRAM Block" call was global, that is, you would call this once for the entire collection of DAC boards being used, and they would all run SRAM in parallel. It looks like this was changed at some point, so that now the server wants a channel identifier (that's the extra `(ss)`) to be passed and expects you to call "New SRAM Block" once for each Microwave and Analog channel (which playback SRAM data). I don't know when this change was made, but it should be fairly straightforward to modify your code to deal with this. You'll likely need to bring the `new_sram_block` call inside a for loop that is looping over channels anyway, and then include the tuple `(device_name, channel_name)` as an additional argument. You're probably passing this channel identifier tuple as the first argument to the functions like `sram_iq_data` where you upload the data, so it should be easy to make the modification.

Coxious, 2015-12-11:
Thanks for your reply, I'll try that.

Coxious, 2015-12-13:
Well I successfully managed to adapted our code to new QubitServer. @pomalley Our test shows that the Latest Java Version is compatible with the old FPGA board, however the latest scala version need new DAC calibration server, which is currently still not compatible.

pomalley, 2015-12-14:
@Coxious great! A few questions:

1. Do you still need the older version of the qubit server (as in #295)? I don't know that we still have it around and it seems like you got it working.
2. What do you mean exactly that the DAC calibration server is not compatible? Not compatible with your code?
3. Is there some documentation we could add to the qubit server, FPGA server, or DAC calibration server to make this easier in the future?

Coxious, 2015-12-17:
Apologize for not replying in time. In last few days I was working on my course assignment and didn't go to the lab.

1. I would be very appreciate if I could get the older version of the QubitServer so that I could learn what have been changed to prevent any potential bugs, however since the latest java version is already functional, it won't be necessary. 

2. Sorry I haven't deep in to the calibration server yet, since I was working on my course assignment (damn it !). I will let you know as soon as I got any progress.

3. Actually I hope I could found some examples for using those servers. This time the biggest difficult is that I do not actually know the relations between each functions of the server. For example,  `new_sram_block` must be called before calling `sram_iq_data_fourier`. This cause a bug that makes me confused quite a while. Only after I carefully read the code of QubitServer do I realize that. And I think pure words documentation might not be able to describe these kind of details, an example might be able to do the work.

closed 2023-04-30 by SchineCompton