labrad/servers #289: Reading timing information from the preamp cards with GHz DACs

amopremcak, 2015-12-11:
We recently upgraded to the latest version of the ghz_fpga_server and are unable to run our measurement code. The server reports an error:
'DAC data readback not supported'
when we attempt to obtain preamp timing information from the DACs. We are running our measurements with DAC build 8. It seems that this functionality was intentionally removed. What was the reason for removing this? What steps should we take to restore this functionality in the server?

DanielSank, 2015-12-11:
That function was removed because the last few version of the DAC firmware don't support it. It probably wouldn't be hard to add that code back in though. Git is your friend here.

amopremcak, 2015-12-11:
The history of this file dates back to Nov 18, 2015 on GitHub, although it is certainly older than that. How can I view changes made to this file before this date?

DanielSank, 2015-12-11:
Before we go down this road, why do you need the timers?

gribeill, 2015-12-11:
We use the preamp cards to read out photon counters (phase qubit style!). Alex, we should be able to pull the relevant code from our old `ghz_fpga_server.py, v4.0`, and `dac.py`.

DanielSank, 2015-12-11:
I see. Well, I guess we pulled the rug out from under you guys. The board interface we're working with now is pretty different from the one you're using; we don't have the "memory" system in the boards any more, i.e. no communication between the boards and the DC rack. I'm sure you guys can figure it out though, especially if you have something to use as a reference. If you need help understanding anything let us know.

amopremcak, 2015-12-11:
Sounds good, thanks.

gribeill, 2015-12-11:
Thanks! Looking over the old and new code I think I see what changes need to be made. We're sticking with DAC build 8 for now, as it's stable for us -- at some point I'm sure we'll upgrade some boards to the latest DAC builds. The jump table functionality is pretty slick.

DanielSank, 2015-12-11:
@gribeill The jump table is nice but beware that we had to ditch the fiber input and output op codes to get it into memory. It's actually probably possible to re-implement the fiber commands as jump table op codes, but we're not going to do it so you'd have to learn a bit of FPGA programming (which is not so bad, you just have to lock the door to your office for a while and think like an FPGA).

DanielSank, 2015-12-11:
One more thing @gribeill and @amopremcak: If you can get the proper support for Build 8 into the FPGA server without breaking backwards compatibility, that would be really nice, and I'd ask that you file a pull request for it. I did a lot of work a few years ago to factor the particulars of how each build works into build-specific classes precisely to make things like this easier.

commit 634f4980 (2016-01-05): Allow reading back DAC timing data as well as ADC demod/avg data.

Fixes #289