labrad/servers #290: Possible bugs in dc_rack_server

amopremcak, 2015-12-11:
1) In the changeLEDs function there appears to be a random write([1L]) statement. Does this serve any purpose? In any case, it is missing a yield in front of it. 

    @inlineCallbacks
    def changeLEDs(self, data):
        """Sets LED status."""
        if isinstance(data, tuple):
            data = 4*data[0] + 2*data[1] + 1*data[2]
        else:
            data &= 0x7
        self.write([1L])
        yield self.write([OP_LEDS | data])
        returnValue(data)

2) In the list_cards function there is a returnValue in place of what should be a return:

    @setting(565, 'list_cards')
    def list_cards(self, c):
        """List cards configured in the registry (does not query cards directly)."""
        dev = self.selectedDevice(c)
        cards = dev.returnCardList()
        returnValue(cards)

DanielSank, 2015-12-11:
1. Definitely a bug. There should be a yield there. Missing yields are all over the dc rack code.

2. Yeah. I don't know why but the dc rack code has a lot of mistakes regarding asynchrony.

amopremcak, 2015-12-11:
Sure thing.