labrad/servers #292: How should forked repos contribute to this one?

DanielSank, 2015-12-11:
Proposed method

1. Initial state

    ```
    our repo
    A <-- B <-- C
                ^
              master

    their repo
    A <-- B <-- C
                ^
              master
```

1. They do some work on a feature branch

    ```
    their repo
    A <-- B <-- C <-- D
                ^     ^
           master     feature
```

1. They submit `feature` as a pull request. It gets reviewed and eventually accepted. They then pull down master from us.

maffoo, 2015-12-15:
@DanielSank, this seems fine. Another way to state it is that we treat PRs from forked repos exactly like our own PRs, except that someone with push rights to this repo will have to do the rebase/merge to master/push, rather than the person who created the PR. We already have a provision for hotfixes where a person other than the author of the change can merge and push, we would just extend that to PRs from external contributors as well.

DanielSank, 2015-12-15:
@maffoo I agree, from our end nothing much differs. It's less clear to me how the downstream users should expect things to work. Any thoughts there?

maffoo, 2015-12-15:
With the way we manage our branches, we're only going to be able to accept patches from people if those patches are based on recent master (or easily rebased on recent master, since that's what we do before actually merging changes). Whether that means they stay close to our master branch or just rebase their own changes on master before making a PR or whatever, that's really up to them.

Remember that git is a distributed system. Every clone is a fully-functioning repo, no different from a "fork" on github. So the kind of branch management we do with our own local changes is really no different from how we should work with outside people. If people in our own group try to manage their branches differently, there's not much we can do to stop that (other than, say, social pressure...), but they're going to have a hard time getting their changes in to master.

maffoo, 2015-12-15:
In other words, yes, I think we should also encourage outside people to stay close to master.

DanielSank, 2015-12-15:
> If people in our own group try to manage their branches differently, there's not much we can do to stop that

I should have said this at the beginning, but my goal here was to generate a paragraph to stick in the contribution guidelines for other groups to use. A bit of good advice can be really helpful, so I would like to make a recommendation on how folks might want to manage their branches.

It sounds like that recommendation is "stick close to master, PR often when you make changes, and if you do anything else you're on your own". How's that sound?

maffoo, 2015-12-15:
That sounds reasonable.

jwenner, 2015-12-15:
Do we even have a contributing doc in this repo (or even a link to one in the readme)? It looks like not (although this should probably be fixed). Maybe a link to the labrad one?

DanielSank, 2015-12-15:
@jwenner yeah linking to LabRAD is probably the right idea.

amopremcak, 2016-01-27:
By "stick close to the master", you guys mean continually merging the master branch into your feature branch correct?

DanielSank, 2016-01-27:
1. Update your forked master regularly.

1. Keep your feature branches close to your master. Get them merged asap, and if a branch is living longer, then you need to either merge master into the feature branch or rebase the feature on top of master. I use the rebase strategy, which has some tricky consequences if you work on multiple computers. I'm not sure if there are undue consequences of the merge strategy.

jwenner, 2016-01-27:
A nice thing about using rebase is that you don't get merge commits every time you rebase (but you do get them each time you merge).

DanielSank, 2016-01-27:
Right, but it could be that you can rebase away the merge commits later :D

maffoo, 2016-01-27:
@DanielSank, in general you cannot rebase away merge commits. That way lies madness. You can take a diff of your branch against master, and just treat that diff like one commit to be applied to master to keep the history linear, and we've had to resort to that in a few cases where master was merged into feature branches and then we wanted to bring the branch back to master, but generally your life will be better if you don't try to mix rebase and merge strategies.