labrad/servers #297: hdf5 datavault error

christopherquintana, 2015-12-15:
Error occurred during a long scan using the datavault from the hdf5 branch. This same error has occurred once or twice before. Unable to access datasets without restarting datavault.

Traceback from scan attempting to add to dataset:

-> 1476             dataset.add([iteration] + [prob[1]] + [nextTime])
   1477             del alg
   1478 

/usr/local/google/home/bryanchen/src/pyle/pyle/datasaver.pyc in add(self, data)
    129             self._create() # make sure the dataset has been created
    130         if len(self.requests) >= self.delay:
--> 131             result = self.requests.pop(0).wait()
    132             self._handle_request(result)
    133         data = np.asarray(data)

/usr/local/google/home/bryanchen/src/pylabrad/labrad/backend.pyc in wait(self)
    405                 # If any Future in the queue fails,
    406                 # we immediately bail.
--> 407                 result.raiseException()
    408             else:
    409                 for func, args, kw in f.callbacks:

/usr/local/google/home/bryanchen/.virtualenvs/pyle/lib/python2.7/site-packages/twisted/python/failure.pyc in raiseException(self)

Error: (0) [Data Vault] Remote Traceback (most recent call last):
  File "/usr/local/labrad/pylabrad/labrad/server.py", line 281, in handleRequest
    result = yield setting.handleRequest(self, c.data, data)
  File "/usr/local/labrad/servers-hdf5_data/datavault/server.py", line 259, in add
    dataset.addData(rec_data)
  File "/usr/local/labrad/servers-hdf5_data/datavault/__init__.py", line 424, in addData
    self.data.addData(data)
  File "/usr/local/labrad/servers-hdf5_data/datavault/backend.py", line 783, in addData
    self.dataset[old_rows:(old_rows + new_rows)] = data
  File "_objects.pyx", line 54, in h5py._objects.with_phil.wrapper (/tmp/pip-build-OsLoUX/h5py/h5py/_objects.c:2316)
    
  File "_objects.pyx", line 55, in h5py._objects.with_phil.wrapper (/tmp/pip-build-OsLoUX/h5py/h5py/_objects.c:2255)
    
  File "/usr/lib64/python2.7/site-packages/h5py/_hl/dataset.py", line 586, in __setitem__
    self.id.write(mspace, fspace, val, mtype)
  File "_objects.pyx", line 54, in h5py._objects.with_phil.wrapper (/tmp/pip-build-OsLoUX/h5py/h5py/_objects.c:2316)
    
  File "_objects.pyx", line 55, in h5py._objects.with_phil.wrapper (/tmp/pip-build-OsLoUX/h5py/h5py/_objects.c:2255)
    
  File "h5d.pyx", line 217, in h5py.h5d.DatasetID.write (/tmp/pip-build-OsLoUX/h5py/h5py/h5d.c:3116)
    
  File "_proxy.pyx", line 128, in h5py._proxy.dset_rw (/tmp/pip-build-OsLoUX/h5py/h5py/_proxy.c:1533)
    
  File "_proxy.pyx", line 354, in h5py._proxy.needs_proxy (/tmp/pip-build-OsLoUX/h5py/h5py/_proxy.c:3242)
    
exceptions.RuntimeError: Unable register datatype atom (Can't insert duplicate key)
 [payload=None]

When I attempt to open a dataset through the Grapher, the following error message occurs:

[image]

maffoo, 2015-12-15:
A few questions to understand a bit better what's happening: did you see the error in the scan first? Did the grapher error happen after that first scan error? Was the grapher error triggered when opening the same dataset as had the scan error, or a different one? Were you able to open any other datasets with the grapher, or did they all exhibit the same error?

christopherquintana, 2015-12-15:
The error in the scan happened first. The grapher error only happened when I tried to view the data in the morning. The grapher error occurred for all datasets that I tried to open, not just the most recent one.

ejeffrey, 2015-12-15:
So I was able to find some suggestions search this online, that the HDF
library may be exhausting identifiers for type tags, since we create new
compound typetags for every dataset.  I am going to try to reproduce this
outside the data vault.

closed 2018-06-25 by christopherquintana