labrad/servers #307: Datavault Memory Error Question

amopremcak, 2016-01-15:
I am investigating the performance of the various data vault settings that are available for adding to a data set. I am receiving an error message when using the add_ex setting from the data vault server. The script is as follows:

import labrad
cxn = labrad.connect()
dv = cxn.data_vault

dv.new_ex("mockBiasSweep", [("Bias1", [1], "v", "V"),("Bias2", [2], "v", "V")], [("Q", "Q(V1,V2)", [1000], "v", "V")])

b = []
for ii in range(0, 1000):
   b.append(ii)
a = []
numSlices =1000
for jj in range(0, numSlices):
   a.append(([jj], [0, 1], b))
dv.add_ex(a)

When I run this I receive the following error message:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\labrad\client.py", line 61, in __call__
    flat = T.flatten(args, tag)
  File "C:\Python27\lib\site-packages\labrad\types.py", line 337, in flatten
    s, t = _flatten_to(obj, types, endianness)
  File "C:\Python27\lib\site-packages\labrad\types.py", line 377, in _flatten_to

    raise FlatteningError(obj, t)
  File "C:\Python27\lib\site-packages\labrad\types.py", line 1558, in __init__
    self.msg = 'Could not flatten %r to %s.' % (data, types)
MemoryError

When I set the variable numSlices = 100, I can add/get (add_ex/get_ex rather) this data set just fine. Is this type of behavior expected when attempting to write a data set of this size all at once? I attempted to use the add_ex_t setting which claims better performance, but I am confused as to how it would like me to format my data. The windows task monitor indicates that I am using only ~50% of the available memory on my machine. I am using fairly recent versions (about a month old) of pylabrad, the scala labrad manager, and data vault server.

DanielSank, 2016-01-15:
I edited your post to put in formatted code. Please check it to make sure I didn't mess anything up.

amopremcak, 2016-01-16:
Looks good to me. I'll take a look for future reference.

amopremcak, 2016-01-18:
After doing some more testing, it appears that I cannot duplicate this issue on my machine at home. This is an issue on our end. I should have done more testing before posting this question.

maffoo, 2016-01-26:
Notice that your memory error is happening locally when trying to flatten data to send to the server, not in the server itself. Also, the memory error is happening when trying to construct the error message, probably because `data` is pretty big (a 1000-element list where each entry is a tuple whose last element is a 1000-element list, even bigger if you increase `numSlices` to 100,000). The error message tries to include a `repr` of this data due to the `%r` formatting code, and that `repr` would be millions of characters long, so I presume the MemoryError is happening when python tries to allocate a big enough string to contain it.

I would suggest putting the call to `add_ex` inside the loop, rather than creating one giant data structure and trying to add the data all at once; that's typically how data-acquisition works anyway.

Also, we might want to change that FlatteningError code to not include a repr of the data it's trying to flatten, or at least to truncate the repr somehow to avoid problems like this (though not sure if that's even possible to do in a general way since you'd still have to construct the full repr before you can truncate it...).

This seems to be one of those cases where an error raised by the error handling code masked the real problem. C'est la vie.

amopremcak, 2016-01-27:
Thanks for looking at this @maffoo. I wrote this script for timing the various settings that the data vault has for appending to a dataset, hence its pathological nature. If I use add_ex (add_ex_t) inside a loop to get around this issue, then I am forced to use get_ex (get_ex_t) inside a loop as well or else it crashes when attempting to return a large dataset all at once. Would you recommend just calling get_ex with a limit, say limit =1, until you exhaust the data set? I know that by default get_ex only returns new data within a particular context, but if you open up an old dataset it will return everything unless you tell it not to.

What I am really interested in is timing performance. I wrote a grapher for browsing datasets and I noticed that it was slowing down substantially when the datasets became large. Any suggestions on how to speed things up would be greatly appreciated.

maffoo, 2016-01-27:
@amopremcak, yes, it's best to call `get*` with a limit and fetch the dataset in smaller chunk. Right now the limit is specified in terms of number of rows, so the number of rows you want per chunk will depend a bit on how large each row is; with one column containing 1000-element arrays, you probably want fewer rows per chunk. We might want to allow specifying a limit in terms of number of bytes instead, which would make it easier to do the chunking appropriately for any dataset.