labrad/servers #312: classmethods in dac.py and adc.py are used improperly

maffoo, 2016-01-28:
There are a number of class methods defined in dac.py and adc.py which are called as instance methods in all cases. In at least one [case](https://github.com/martinisgroup/servers/blob/master/fpgalib/adc.py#L444), the call also incorrectly passes a reference to the object as a first parameter, even though a class method would expect a class there (and the class parameter gets added automatically anyway, just like the self reference gets added automatically when calling regular instance methods). We should clean this up.

DanielSank, 2016-01-28:
I see that the invocations of `regRun` are plain wrong, but what's wrong with a class method being called from instances in all cases? I used class methods wherever the method does not need to know anything about the state of any instance.

maffoo, 2016-01-28:
Yeah, I guess you're right. It seemed strange to me to call a class method through an instance, but it certainly works and is even called out as OK in the docs for `classmethod` (https://docs.python.org/2/library/functions.html#classmethod). So I guess we just need to fix up the call site(s) that pass extra args.

DanielSank, 2016-01-28:
>  but it certainly works and is even called out as OK in the docs for classmethod

Yeah, I think this is a pretty standard way to use it.