labrad/servers #313: telecommServer question

amopremcak, 2016-02-04:
I am messing around with the telecommServer but I cannot figure out how to get it to send an sms message. The registry seems to be looking for an ```smsUsers``` key of the form: 
```[("GROUPMEMBERNAME", "8008675309")]``` 
but this is failing to send a text. I am able to get the ```send_mail``` setting to work just fine. Any tips?

DanielSank, 2016-02-04:
Most cell phone providers have a way to send texts via email. In the registry, you've got to have something like this at `Servers/Telecomm`:

```
domain -> "physics.ucsb.edu"
smsUsers -> [("DANIEL","blahblah@txt.att.net"), etc.]
smtpServer -> "smtp.physics.ucsb.edu"
```

DanielSank, 2016-02-04:
It would be dope if you could add a note explaining that to the server code so people know what to do :/

amopremcak, 2016-02-04:
Sure thing. Should I just add this as some comments within the telecommServer and submit a pull request?

DanielSank, 2016-02-04:
Yep.

amopremcak, 2016-02-04:
Agreed, thanks for writing this and for the feedback @DanielSank. My departments smtp server isn't allowing me to send texts via email because the addresses don't end with the schools domain name (@wisc.edu). I'll do some hacking around tomorrow, compile a list of some common cellphone providers and add some notes to this server within the next day or so.

pomalley, 2016-02-04:
Ah, yes, our SMTP server was administered... loosely... for on-campus IP's, which got one of us a personal visit from the admin when someone (somehow...) started getting cat facts.

amopremcak, 2016-02-04:
Lol

DanielSank, 2016-02-04:
@pomalley it's spelled "Catfax" and I still have the code if you're interested.

amopremcak, 2016-02-06:
Before I submit a PR I have a quick question. Are you guys able to send out sms messages without providing login credentials? I've got the server working just fine but I have to provide login credentials when I send messages outside of the schools domain name. Also I find that I am having to call the ```SMTP.starttls()``` method to get this to work. By the looks of the server, things are probably working just fine on your end.

DanielSank, 2016-02-06:
> Are you guys able to send out sms messages without providing login credentials?

Yes, because a machine inside the physics department at Broida doesn't need to authenticate to use the smtp server. Please do add the authentication bits though, and allow them to be bypassed if desired.

amopremcak, 2016-02-06:
Sounds good to me.

DanielSank, 2016-12-15:
@amopremcak any progress here?

amopremcak, 2016-12-27:
I submitted a new pull request to you guys just now @DanielSank. However I just remembered that I had already submitted a pull request for this exact server and was waiting on your feedback to determine whether or not the password should be placed in the registry or not.

DanielSank, 2017-01-04:
Apologies for the delay. Apparently I'm not getting notifications from this repo. Can you link to your previous PR. I'll take a look.