labrad/servers #314: GHzDAC Bringup Questions

Brian-Ballard, 2016-02-05:
I am trying to bring up some new GHzDACS using the most recent GHz_fpga_server and GHz_dacs_bringup and Firmware build 15.  I previously used very old software for bringup. 

Firstly: I want to know if the FIFO failure message is actually problematic as i have not seen it before. 

Secondly: Is there a current equivalent to the serializer_test or function call that mimics it's functionality (creating simple patterns on all serializer outputs)?

jwenner, 2016-02-08:
FIFO failures can indeed be problematic and can, depending on the frequency, be a symptom of a more serious failure. In particular, they can be indicative of getting a time offset between DAC A and DAC B which can appear/disappear each time you run the bringup script. Note that this is true even for builds 7/8.

If you are getting failures at the 5-10% level, this seems to be normal (although still disturbing). In this case, there only seems to be a time offset when you get FIFO failures. Hence, it should be OK at least for now to rerun the bringup script whenever you get these failures. However, it would still be good to check that you are using the correct SOF/POF.

If you are getting too many more failures, this could be indicative of using an improper SOF/POF file for the board. If this is the case, you can get the time offset (dis)appearing independent of FIFO failures. Hence, the only solution in this case is to use other SOF/POF files.

To determine the correct SOF/POF files, we have scripts written which scan over multiple SOFs and display the results. However, I'm waiting for #315 to be approved before uploading them to this repository so that there is a home for them. (They are currently in a private repository for the Martinis group.)

In the meantime, could you please give me an idea as to how frequently you are getting FIFO failures? Also, what SOF/POF files do you have access to for Build 15?

jwenner, 2016-02-08:
For the serializer test, we do have such a script, although I don't know if it has been updated to work with the jump table boards. (@afowler, @JulianSKelly, @pomalley, @jrainbo, do any of you know?) However, like the scripts for scanning over SOF/POF files, it is currently in a private repository.

Brian-Ballard, 2016-02-08:
So far I have only tested one board.  I am getting fifo errors on every (~10) pass of reinitialization.

I used the file  GHzDAC_V9_B15_P0_P0_s1.sof  in martinisgroup/hardware/fpga_src/ghzdac/GHzDAC_V9_B15/

jwenner, 2016-02-09:
As all the SOF/POF files for Build 15 are in that directory, it looks like you have access to them.

Brian-Ballard, 2016-02-10:
I am not worried about the POF/SOF scanning scripts.  I would however like the serializer script if it compatable.

jwenner, 2016-02-11:
Could you please let me know your email address so I can email you these scripts? My email address is on the Martinis group website.

Be aware that the serializer script may or may not work on this jump table build, so _caveat emptor_.

closed 2016-02-11 by Brian-Ballard

Brian-Ballard, 2016-02-11:
Closing notes: 
Serializer test does not work with build 15.  Does work with build 8.

Ghz dacs bringup works only partially for each build.  
Pll Init and Bringup:  Both
Static Dac Outputs (functions 1-4): Only on build 8.  Mapping is old mapping so does not match script. (only functions 1-3 work with 3 being min 2 max and 1 mid range) On build 15 script crashes.
Sine wave generation (functions 5-7): Both
Cross Controller functions (-,+,*,/): only on build 15.  With Build 8 sending a cross controller change turns off sine output.