labrad/servers #317: Qubit server timeout - 10 minutes?

jwenner, 2016-02-09:
Is there a reason the timeout for the qubit sequencer server is made 10 minutes instead of a perhaps more reasonable time (e.g., 1 minute)? Running Qubit Sequencer 0.7.1.

Error message below. The only noteworthy message I see is the timeout, but there may be other errors I can't see.
```
In [26]: cr.prob_shaped_release(s,'qB','Bc','Bf',cuttimes=st.r[2:600:2,ns],stats
=3000,length=500*ns,tau=200*ns,t1min=92*ns,noisy=True)
---------------------------------------------------------------------------
Error                                     Traceback (most recent call last)
<ipython-input-26-895688b9b11d> in <module>()
----> 1 cr.prob_shaped_release(s,'qB','Bc','Bf',cuttimes=st.r[2:600:2,ns],stats=
3000,length=500*ns,tau=200*ns,t1min=92*ns,noisy=True)

C:\pyle\pyle\dataking\alice_bob_res\capture_release.pyc in prob_shaped_release(S
ample, measure, coupbias, freqbias, length, tau, t1min, t1i, cuttimes, name, sta
ts, save, noisy, collect)
    432
    433     data = sweeps.grid(func, axes, dataset=dataset, save=save, collect=c
ollect,
--> 434                        noisy=noisy)
    435     if collect:
    436         return data

C:\pyle\pyle\dataking\sweeps.pyc in grid(func, axes, dataset, **kw)
    320         dataset=dataset,
    321         abortPrefix=[1],
--> 322         **kw
    323     )
    324

C:\pyle\pyle\dataking\sweeps.pyc in _run(func, sweep, save, dataset, abortable,
abortPrefix, collect, noisy, pipesize)
    224                                            params, name, **kwargs)
    225     else:
--> 226         for _ in it: pass
    227
    228 def gridSweep(axes, unit_strs):

C:\pyle\pyle\datasaver.pyc in capture(self, iterable)
    146         """
    147         with self as dataset:
--> 148             for data in iterable:
    149                 dataset.add(data)
    150                 yield data

C:\pyle\pyle\pipeline.pyc in gen()
     99             runners.append(runner)
    100             if len(runners) >= size:
--> 101                 done, result = advance()
    102                 if done:
    103                     yield result

C:\pyle\pyle\pipeline.pyc in advance()
     80             # loop over runners, advancing until one completes
     81             for runner in itertools.cycle(runners):
---> 82                 runner.advance()
     83                 if runner.done:
     84                     break

C:\pyle\pyle\pipeline.pyc in advance(self)
    123         if not self.done:
    124             try:
--> 125                 self.gen.next()
    126             except _DefGen_Return, e:
    127                 self.done = True

C:\pyle\pyle\pipeline.pyc in _run(gen)
    173             elif is_generator(result):
    174                 try:
--> 175                     for _ in _run(result):
    176                         yield
    177                     result = None

C:\pyle\pyle\pipeline.pyc in _run(gen)
    173             elif is_generator(result):
    174                 try:
--> 175                     for _ in _run(result):
    176                         yield
    177                     result = None

C:\pyle\pyle\pipeline.pyc in _run(gen)
    170                     result = result.wait()
    171                 except Exception, e:
--> 172                     result = gen.throw(e)
    173             elif is_generator(result):
    174                 try:

C:\pyle\pyle\dataking\alice_bob_res\capture_release.pyc in func(cuttime)
    427         alg.compile(correctXtalkZ=False)
    428
--> 429         result = yield Sample.run(alg, stats, dataFormat='iqRaw')
    430         probs = readout.IQToProbs(result, alg.qubits, measure)
    431         returnValue([np.squeeze(probs)[1]])

Error: (0) java.util.concurrent.TimeoutException: Futures timed out after [10 mi
nutes]
        at scala.concurrent.impl.Promise$DefaultPromise.ready(Promise.scala:219)

        at scala.concurrent.impl.Promise$DefaultPromise.result(Promise.scala:223
)
        at scala.concurrent.Await$$anonfun$result$1.apply(package.scala:190)
        at scala.concurrent.impl.ExecutionContextImpl$DefaultThreadFactory$$anon
$2$$anon$4.block(ExecutionContextImpl.scala:48)
        at scala.concurrent.forkjoin.ForkJoinPool.managedBlock(ForkJoinPool.java
:3640)
        at scala.concurrent.impl.ExecutionContextImpl$DefaultThreadFactory$$anon
$2.blockOn(ExecutionContextImpl.scala:45)
        at scala.concurrent.Await$.result(package.scala:190)
        at org.labrad.qubits.QubitContext.run_experiment(QubitContext.scala:1064
)
        at sun.reflect.GeneratedMethodAccessor21.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at scala.reflect.runtime.JavaMirrors$JavaMirror$JavaVanillaMethodMirror1
.jinvokeraw(JavaMirrors.scala:373)
        at scala.reflect.runtime.JavaMirrors$JavaMirror$JavaMethodMirror.jinvoke
(JavaMirrors.scala:339)
        at scala.reflect.runtime.JavaMirrors$JavaMirror$JavaVanillaMethodMirror.
apply(JavaMirrors.scala:355)
        at org.labrad.SettingHandler$$anonfun$org$labrad$SettingHandler$$invoke$
1.apply(SettingHandler.scala:233)
        at org.labrad.SettingHandler$$anonfun$org$labrad$SettingHandler$$invoke$
1.apply(SettingHandler.scala:231)
        at scala.Function1$$anonfun$andThen$1.apply(Function1.scala:52)
        at scala.Function1$$anonfun$andThen$1.apply(Function1.scala:52)
        at org.labrad.SettingHandler$$anon$1.apply(SettingHandler.scala:243)
        at org.labrad.SettingHandler$$anon$1.apply(SettingHandler.scala:241)
        at org.labrad.Reflect$$anonfun$15$$anonfun$apply$4$$anonfun$apply$5.appl
y(reflect.scala:53)
        at org.labrad.Reflect$$anonfun$15$$anonfun$apply$4$$anonfun$apply$5.appl
y(reflect.scala:53)
        at scala.Option.map(Option.scala:146)
        at org.labrad.Reflect$$anonfun$15$$anonfun$apply$4.apply(reflect.scala:5
3)
        at org.labrad.Reflect$$anonfun$15$$anonfun$apply$4.apply(reflect.scala:5
2)
        at org.labrad.Server$$anonfun$org$labrad$Server$$handleRequest$1$$anonfu
n$apply$7.apply(Server.scala:181)
        at org.labrad.Server$$anonfun$org$labrad$Server$$handleRequest$1$$anonfu
n$apply$7.apply(Server.scala:166)
        at org.labrad.Server$.handle(Server.scala:225)
        at org.labrad.Server$$anonfun$org$labrad$Server$$handleRequest$1.apply(S
erver.scala:166)
        at org.labrad.Server$$anonfun$org$labrad$Server$$handleRequest$1.apply(S
erver.scala:166)
        at org.labrad.util.AsyncSemaphore$$anonfun$map$1.apply(AsyncSemaphore.sc
ala:83)
        at org.labrad.util.AsyncSemaphore$$anonfun$map$1.apply(AsyncSemaphore.sc
ala:82)
        at scala.util.Success$$anonfun$map$1.apply(Try.scala:237)
        at scala.util.Try$.apply(Try.scala:192)
        at scala.util.Success.map(Try.scala:237)
        at scala.concurrent.Future$$anonfun$map$1.apply(Future.scala:235)
        at scala.concurrent.Future$$anonfun$map$1.apply(Future.scala:235)
        at scala.concurrent.impl.CallbackRunnable.run(Promise.scala:32)
        at scala.concurrent.impl.ExecutionContextImpl$AdaptedForkJoinTask.exec(E
xecutionContextImpl.scala:121)
        at scala.concurrent.forkjoin.ForkJoinTask.doExec(ForkJoinTask.java:260)
        at scala.concurrent.forkjoin.ForkJoinPool$WorkQueue.runTask(ForkJoinPool
.java:1339)
        at scala.concurrent.forkjoin.ForkJoinPool.runWorker(ForkJoinPool.java:19
79)
        at scala.concurrent.forkjoin.ForkJoinWorkerThread.run(ForkJoinWorkerThre
ad.java:107)
 [payload=None]

In [27]:
```

maffoo, 2016-02-09:
No good reason for this; I set the timeout longer than I thought any individual data point might take (including in the presence of load from other users). Might be better to make this variable with a shorter default.

pomalley, 2016-02-09:
How long do the RRTO "data points" take? (IIRC one run is one data point with many measures, but I could be wrong.)

btchiaro, 2016-02-09:
Hor the RRTO 1e6 points at a sampling interval of 120 usec is 2 min of pure
acquisition time plus overhead.  I'm not sure how much "overhead" is.

ejeffrey, 2016-02-09:
I think the 10 min timeout is unfortunate but necessary due to issues like
the RRTO and the surface code that may need to run for minutes.

What we need to fix here is to allow data to be streamed back in real time
(even if the data analysis doesn't run until it is over).  Then we could
reduce the timeout considerably.

One other possibility if you just want to get faster error recovery is to
set a timeout on the direct ethernet server to fail out if no packets are
received for X seconds.

Evan

btchiaro, 2016-02-09:
I believe that I've had problems in the past by taking large numbers of
RRTO points and overloading the DE server buffer. It seems like live
streaming the data could fix that problem too.  I like that idea.