labrad/servers #318: Scans fail when stats are too high

zchen088, 2016-02-16:
Scans tend to timeout when the stats is of order a few tens of thousand. This problem is compounded by #317.

Interestingly, RTO experiments seem to be able to return up to a million demods without any problems. Does anyone know why there is this discrepancy?

@maffoo @ejeffrey

ejeffrey, 2016-02-16:
I believe the "number of starts" field in the FPGA run packet is a 16 bit
integer, and therefore limited to 65535 stats.  The RTO uses the jump table
/ trigger table to collect the multiple stats, so it doesn't have this
limitation.

pomalley, 2016-02-16:
In my chemistry experiment, I think I was able to do up to 30,000 stats,
but it wasn't too reliable past 10 or 15 thousand.

zchen088, 2016-02-16:
So it sounds like we should add a check in the FPGA server to see if stats exceeds max stats, but I also agree with @pomalley that things in general seem less reliable with high stats.

ejeffrey, 2016-02-16:
Yes, I think what happens is that the FPGA server truncates the stats field
to the low 16 bits when writing to the DAC, but then tells the direct
ethernet server to wait for N packets as if it wasn't truncated, causing a
timeout.

There might also be a signed/unsigned error reducing the actually usable
stats to 32767, but I am pretty sure in the past I have been able to run
50,000 stats.

Can you be more specific about what is unreliable above 10,000 stats?  I
know it gets slow.  Are you just getting timeout errors, or is something
else going on?

Evan

zchen088, 2016-02-16:
I'll see if I can characterize it better.

zchen088, 2016-02-16:
Ok, I ran with 65000 stats 20 times and it ran ok. 66000 stats causes a timeout, which agrees with what @ejeffrey said.

DanielSank, 2016-02-21:
Cool, then we know what to fix.