labrad/servers #319: Storing integers in the datavault

zchen088, 2016-02-16:
My scans fail when I try put arrays of integers in the datavault. Is this a limit with the datavault, i.e. it is not possible to store integers in the datavault, or is it a bug with the datasaver?

@ejeffrey

ejeffrey, 2016-02-16:
So the datavault can only store floats.  Also, the type tag for
datavault.save is supposed to be *2v, so pylabrad should be forcibly
converting to floats.  I don't know why this would be happening.

Evan

maffoo, 2016-02-16:
@zchen088, can you provide a stack trace for the error? Among other things, we need to know if there's a problem on your side (while preparing a request to the datavault) or in the datavault itself, after the request has been sent.

ejeffrey, 2016-02-16:
The problem here is that types.py knows how to flatten integers to *v, and
lists of integers to *v, but it can't flatten a numpy array of integers to
*v, so this is a pylabrad bug.

zchen088, 2016-02-16:
Ok, will raise issue in pylabrad, closing for now.

closed 2016-02-16 by zchen088