labrad/servers #320: Slaving DAC board

rpbarnes, 2016-02-19:
I need to slave the DAC board to an external pulse generator. Which pin on the 'daisy up' ethernet plug do I need to raise? Is the logic level 5 V? and what are the constraints on trigger length?

Thanks for the help!
~Ryan

ejeffrey, 2016-02-19:
What do you need this to do?  If you need accurate timing, I think this is
going to be really hard.

The daisy chain pinout is on page 7 of this document:
https://matrix-reloaded.physics.ucsb.edu/twiki/pub/Electronics/GHzDAC%28FPGA%29/GHzDACv2_2.pdf

I believe you want daisy0in+ and daisy0in-, which are pins 4 and 5 on the
DAISY_UP plug.  The other options are pins 7,8 for daisy0in.

The signaling is LVDS.  This means that you have to drive 3.5 milliamp into
either the + or the - input, and connect the complimentary pin to ground.
This will generate ~350 mV across the 100 ohm load resistor shown in the
schematic.

All of this is fine.  The problem I think you will have is that the LVDS
I/Os are synchronous with the onboard FPGA clock at 250 MHz, and the
trigger signal is a single cycle '1' value, i.e., 4 nanosecond pulse.  The
DAC only starts on the FPGA clock, so if your trigger is asynchronous with
the DAC, you will have 4 ns uncertainty in your start time.

I believe, but do not know for sure, that daisy inputs are ignored while
the DAC is running.  This is good news, it means that if your pulse is
longer than 4 ns (which you almost certainly want it to be to make sure you
don't miss it entirely), then you won't cause the board to retrigger.
after being started.  However, we have never tested this.  In any case, I
would try to keep the pulse length short.

If you can live with 4 ns start jitter, you can probably make it work.
Otherwise, you will need to sync your trigger to the DAC clock, presumably
by using clocks derived from the same 10 MHz reference you are using on the
DAC.

Can you do the trigger the other way?  Use one of the ECL outputs on the
DAC to trigger external equipment?  We do this for triggering
oscilloscopes, and it has extremely low jitter.

rpbarnes, 2016-02-19:
I think we can live with a 4 ns jitter in the start time of the DAC waveform.

We can try putting the pulse generator on the same clock as the DAC to maybe circumvent this jitter.

We tried doing this the other way, setting the DAC as master, but the pulse generator that runs the rest of the spectrometer has very limited functionality when slaved and cannot run the spectrometer. Also switching to run the spectrometer off of the DAC is problematic because there aren’t enough independent trigger channels. We’re interested in trying to slave the DAC because it’s currently the simplest option however if this proves troublesome we may rethink our approach.

I will keep you updated with my progress.

Thanks for your reply,
~Ryan