labrad/servers #321: Expose slave mode for individual board execution

DanielSank, 2016-02-21:
It could be useful to run a board slaved to something else. To support this, it would be nice to allow setting a board to run as a slave directly.

btchiaro, 2016-02-21:
+1 to this. That would be really helpful for my RTO calibration
measurements.

DanielSank, 2016-02-21:
@btchiaro how? I've already written the code (see branch 321) but it would help to know how it will be used.

btchiaro, 2016-02-21:
I'd be using a dac instead of the Agilent to output my calibration sin
wave.

DanielSank, 2016-02-21:
Why do you need an explicit setting for slave mode for that?

btchiaro, 2016-02-21:
I latched on to the "individual board execution" in the thread title.  I'm
not actually sure that "explicit slave mode" is what I need.  I need a
source that continuously outputs a sin wave that is independent of the dacs.

DanielSank, 2016-02-21:
We can already do that :)

btchiaro, 2016-02-21:
How do I invoke that functionality?

DanielSank, 2016-02-21:
You get a board that's not in he board group, write some SRAM to it, run it with loop=True, and Bob's your uncle. We already talked about this a while ago but you wanted to go with the Agilent.