labrad/servers #323: dac.py raises jump table error incorrectly

JulianSKelly, 2016-02-25:
[This line](https://github.com/martinisgroup/servers/blob/master/fpgalib/dac.py#L1489) raises an error about ```from_addr``` being too close together on a valid sequence.

The FPGAs have a restriction that jump table op codes cannot be executed more often than every 4 clock cycles, due to pipelining. However, this code is looking for how ```from_addr``` are defined __in memory__, not at what times they are executed. 

I believe the only way to fix this is to emulate the experiment and look for when this may be violated, unless I am mistaken. 

I think the best course of action may be to remove this check altogether for now. Or it might be better to remove this check and move it into pyle for the time being, where we already emulate the jump table to make the retrigger table.

also ```JT_MIN_FROM_ADDR_SPACING = 2``` when it should ```=4```

pomalley, 2016-02-25:
I put this check in there based on the following line from John's documentation:

`FromAdr must be separated by 4 or more from other FromAdr’s (due to pipelining).`

I took this to mean that they had to be separated by at least four in memory. I tested this on the bench but did not distinguish the case where they were adjacent in memory but executed further apart in time. IIRC things still mess up if you put a command right after another one even if they're spaced out (e.g. another command right after an IDLE) but it was a while ago.

Also, I _think_ the reason I put 2 in there is because of that factor of 4 that kept tripping me up (i.e. it should be 4//4 = 1, but needs to be 2 because of offsets). Definitely could be wrong though.

JulianSKelly, 2016-02-25:
I spoke with @johnmartinis about this before posting the issue, and he wrote this with the timing execution in mind, rather than the memory placement.

I am running into this issue not with ```IDLE``` as you comment here (I'm not sure if you can jump within 4 cycles of an ```IDLE``` ending...), but with the following example.

If i have my psuedo jump table below for the following situation. 

I have a pi pulse in sram from 0-32.
I have another pi pulse in sram from 32-64.
I have a measure pulse in memory from 64-1000.
My sequence is pi pulse 1 -> measure -> pi pulse 2 -> measure, re-using the measure sram.
```python
jump_table = [
JUMP(from_addr=32, to_addr=64),
JUMP(from_addr=1000, to_addr=32),
END(from_addr = 1000)
]
```
This exception will rear its head.

Instead, we should have exceptions that look to the next entry, and make sure that one's ```from_addr``` is not within the error time of the current ```from_addr```.

pomalley, 2016-02-25:
OK. It's good that we got this cleared up. This PR should also change the line at the very bottom of JumpTableIntro.md. Did you test this (using "the same" from_addr twice) out on a board?

JulianSKelly, 2016-02-25:
I have not been able to test this on hardware yet.

commit 870a9a5c (2016-02-27): Fix 'from address too close' behavior in dac.py
Improve documentation for jump_table in dac.py
Add test_dac
Change dac build 15 offsets:
*JT_MIN_FROM_ADDR_SPACING -> JT_MIN_CLK_CYCLES_BETWEEN_OPCODES
*JT_MIN_CLK_CYCLES_BETWEEN_OPCODES: 2->4 in line with GHZDAC8 specs
*JT_MIN_FROM_ADDR remove offset
*JT_MIN_END_ADDR remove offset
Fixes #323

Review: @maffoo
commit 2310d0eb (2016-03-09): Fix 'from address too close' behavior in dac.py
Improve documentation for jump_table in dac.py
Add test_dac
Change dac build 15 offsets:
*JT_MIN_FROM_ADDR_SPACING -> JT_MIN_CLK_CYCLES_BETWEEN_OPCODES
*JT_MIN_CLK_CYCLES_BETWEEN_OPCODES: 2->4 in line with GHZDAC8 specs
*JT_MIN_FROM_ADDR remove offset
*JT_MIN_END_ADDR remove offset
Fixes #323
*Add test, error for long jump tables

Review: @maffoo
PR #324 (2016-03-09): Fix unwanted exceptions from dac 15 class.
Fixes #323

Change error checking for when two jump table commands are executed too closely in time, rather than too closely in memory.
closed 2016-03-09 by JulianSKelly