labrad/servers #330: Data Vault accepts infinite values

thchwhite, 2016-03-08:
In the course of writing resonator code I found that the data vault will accept infinite values and save them to the dataset.  This problem arose because it was processed data, but could potentially happen any time someone divides by 0 accidentally.  Do we want to allow this behavior, or should the data vault throw an error at this point?

ejeffrey, 2016-03-08:
> Assigned #330 <https://github.com/martinisgroup/servers/issues/330> to
> @ejeffrey <https://github.com/ejeffrey>.

I think allowing +/- inf and nan in datasets is fine -- they are legal
floating point values for a reason.  I would say that either your
datataking code should not generate them or your analysis code should
handle it properly.  If anyone has a strong reason to object, please
comment.

maffoo, 2016-03-08:
I agree with @ejeffrey. It seems best to have the datavault just store the float values it is given with full fidelity.

ejeffrey, 2016-03-08:
I just talked to Ted about this.  The real problem is that the grapher
can't plot datasets with infinite values, which seems like a grapher bug,
not a datavault problem.  Does the new grapher handle this properly?

As an example of when you might want to store inf is if you are reading a
pressure gauge that is overloaded.  You need to store something, and any
number you store is most likely wrong.  inf tells you that is A) not a
legal, and B) above the legal range (rather than below, or a
missing/defective sensor which would be NaN)

pomalley, 2016-03-08:
Yes, I ran into this problem with the supermonitor; the supermonitor just ignores bad values but it was super annoying that the grapher couldn't handle it because it would have been good for debugging to see the data on the grapher as well. Alas.