labrad/servers #333: Understanding of Daisy Chain

Coxious, 2016-03-12:
Hi,
We are trying to connect a FPGA board we made to current system. While we reading the documents of the daisy chain, we found that for each RJ45 port, there are two differential signals input, daisy0 and daisy1. I guess this signal works like a trigger, synchronizing the collection of boards. But why where are two signals? How does it work? Do we have more detailed documentation about that?

Thanks a lot.

ejeffrey, 2016-03-12:
There are actually 4 signals -- two that go "down" the chain, and two that
go "up".  Only a single channel is actually used, which acts as a trigger
to synchronize multiple boards. The way it works is that after programming
SRAM and other configurations for all the boards, we first send a register
write to all the slave boards to run them in slave mode.  Then we send a
register write to the master board, which starts, and sends a pulse down
the daisy chain.  Each subsequent board receives the start pulse from the
board above it, starts running its sequence, and repeats the start pulse
down the daisy chain.  This allows all the boards to be precisely
synchronized.  I think that is daisy0, but I am not sure.

There is untested support in the latest jump table firmware for sending
data "up" the chain from an ADC board and allowing that to control the jump
table.

Coxious, 2016-03-14:
Thanks for your reply. I've checked and looks daisy0 carries the signal.

closed 2023-04-30 by SchineCompton