labrad/servers #334: timeout bug in gpib server

ypzhong, 2016-03-14:
Hi,

    I tried to use the new labRAD but encountered GPIB timeout frequently. Finally I found that there is a possible bug in the gpib server, it seems that all the upper level configuration of the instrument finally configures the timeout via this function in gpib_server.py:
@setting(2, time='v[s]', returns='v[s]')
    def timeout(self, c, time=None):
        """Get or set the GPIB timeout."""
        if time is not None:
            c['timeout'] = time
        return c['timeout']

However it only updates the context timeout, but never sends out the update to the instrument in reality. This is not a problem when the communication is short because the pyvisa itself has a default timeout already set there. But when we need to wait for half a minute, then it becomes a serious problem.

    The solution should be simple, I changed it in my computer like:

    @setting(2, time='v[s]', returns='v[s]')
    def timeout(self, c, time=None):
        """Get or set the GPIB timeout."""
        if time is not None:
            c['timeout'] = time
            instr = self.getDevice(c)
            instr.timeout = time['ms'] #added by Youpeng. Note that for pyvisa>1.5, timeout is in unit of millisecond. March 14, 2016.
            print c['addr']+' timeout is',time
        return c['timeout']

and it works now.

DanielSank, 2016-12-15:
@ypzhong Thank you for this information. Please note that the whole point of having this code repository public is so that people can submit changes. Could you please submit your change using a github pull request?

ypzhong, 2016-12-16:
@DanielSank  OK. Actually I don't use github much and I haven't learned how to use github pull request yet...... But I will figure it out and get accustomed to it. Thanks.