labrad/servers #342: Datavault tests broken

jwenner, 2016-04-19:
The datavault tests are failing with the following error message, which seems to be saying that `data_vault` can not be found. This is causing test failures in #340 and #341.

```
 test_datavault.test_create_dataset_setup 
('tests/test_datavault.py', 31, 'test_create_dataset')
@pytest.yield_fixture
    def dv():
        with labrad.connect() as cxn:
>           dv = setup_dv(cxn)

servers/tests/test_datavault.py:29: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
servers/tests/test_datavault.py:22: in setup_dv
    dv = cxn.data_vault
pylabrad/labrad/client.py:244: in __getattr__
    return self._attrs[key]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = manager
registry, key = 'data_vault'

    def __getitem__(self, key):
        try:
            return super(DynamicAttrDict, self).__getitem__(key)
        except KeyError:
            # force refresh and try again
            if self._parent:
                self._parent.refresh(now=True)
            try:
                return super(DynamicAttrDict, self).__getitem__(key)
            except KeyError:
>               raise NotFoundError(key)
E               NotFoundError: (10) Could not find "data_vault". [payload=None]

pylabrad/labrad/client.py:144: NotFoundError
```

maffoo, 2016-04-19:
The failure seems to be due to labrad/pylabrad#242. From the datavault error log in one of the failed test runs:
```
Traceback (most recent call last):
  File "servers/data_vault.py", line 45, in <module>
    from datavault.server import DataVault
  File "/opt/TeamCity/buildAgent/work/872fe26a728e05ba/servers/datavault/server.py", line 13, in <module>
    class DataVault(LabradServer):
  File "/opt/TeamCity/buildAgent/work/872fe26a728e05ba/servers/datavault/server.py", line 93, in DataVault
    def cd(self, c, path=None, create=False):
  File "/opt/TeamCity/buildAgent/work/872fe26a728e05ba/pylabrad/labrad/decorators.py", line 44, in decorator
    return Setting(func, lr_ID, lr_name, returns, unflatten, **params)
  File "/opt/TeamCity/buildAgent/work/872fe26a728e05ba/pylabrad/labrad/decorators.py", line 131, in __init__
    "unless other arguments are optional".format(args[0]))
ValueError: First argument path cannot accept '' unless other arguments are optional
```

maffoo, 2016-04-19:
The fix for labrad/pylabrad#242 is in pylabrad 0.95.2. Rerunning the tests revealed another problem; see #343.

jwenner, 2016-05-16:
All tests now pass in master. Thanks, @maffoo.

closed 2016-05-16 by jwenner