labrad/servers #347: Understand the registry key 'boardGroups'

chao-song, 2016-05-13:
Hi all, I'm a student in Haohua's lab. I have a problem when trying to understand the registry key  'boardGroups'. 
In this key, each board name is followed by a number, like '[('DAC 1', 3), ('DAC 0',0)]'. What's the mean of these number? Where do we use them?

ejeffrey, 2016-05-13:
The number in 'DAC 1' is the dip switch setting on the PCB that sets the
MAC address.  The number 3 is a start delay.  The purpose here is to
account for the propagation delay of the daisy chain so that all boards
start at the same time.  The fpga server adds that number to the delay you
specify in your scan.  In some sense it is redundant since you have to
measure cable delays anyway but the idea is that if you remove a board from
the daisy chain and update the board group key your measured delays will
already be very close to correct.

Evan

chao-song, 2016-05-14:
Thank you. But what's the unit of the start delay here? ns?4 ns? or some other unit?

maffoo, 2016-05-14:
The unit is cycles of the FPGA clock which runs at 250 MHz, so each cycle is 4 ns.