labrad/servers #351: Some Device servers must be started after GPIB Device Manager

jwenner, 2016-06-21:
If GPIB device servers are started before the GPIB Device Manager, some servers pick up the corresponding devices while others don't. In particular, the Anritsu and Lakeshore Diode servers recognize their corresponding devices. However, the Hittite and Lakeshore Ruox do not recognize their corresponding devices (so `list_devices` gives an empty list).

In particular, I get the following error message from the servers which do not recognize their devices:
```
2016-06-21 13:02:51-0700 [LabradProtocol,client] GPIB Device Manager now serving

INFO:labrad.lakeshore_ruox:2016-06-21 13:02:51-0700 [-] Unhandled error in Defer
red:
2016-06-21 13:02:51-0700 [-] Unhandled Error
        Traceback (most recent call last):
          File "C:\Python27\lib\site-packages\labrad\protocol.py", line 272, in
_dispatchMessage
            d = func(msgCtx, data, *args, **kw)
          File "C:\Python27\lib\site-packages\labrad\server.py", line 463, in _s
erverConnected
            self.serverConnected(ID, name)
          File "C:\Python27\lib\site-packages\labrad\gpib.py", line 310, in serv
erConnected
            self.connectToDeviceManager()
          File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1
274, in unwindGenerator
            return _inlineCallbacks(None, gen, Deferred())
        --- <exception caught here> ---
          File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1
128, in _inlineCallbacks
            result = g.send(result)
          File "C:\Python27\lib\site-packages\labrad\gpib.py", line 293, in conn
ectToDeviceManager
            manager = self.client[self.deviceManager]
          File "C:\Python27\lib\site-packages\labrad\wrappers.py", line 553, in
__getitem__
            return self.servers[key]
          File "C:\Python27\lib\site-packages\labrad\support.py", line 181, in _
_getitem__
            return dict.__getitem__(self, k)
        exceptions.KeyError: 'GPIB Device Manager'
INFO:labrad.hittite_t2100_server:2016-06-21 13:02:51-0700 [-] Unhandled error in
 Deferred:
2016-06-21 13:02:51-0700 [-] Unhandled Error
        Traceback (most recent call last):
          File "C:\Python27\lib\site-packages\labrad\protocol.py", line 272, in
_dispatchMessage
            d = func(msgCtx, data, *args, **kw)
          File "C:\Python27\lib\site-packages\labrad\server.py", line 463, in _s
erverConnected
            self.serverConnected(ID, name)
          File "C:\Python27\lib\site-packages\labrad\gpib.py", line 310, in serv
erConnected
            self.connectToDeviceManager()
          File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1
274, in unwindGenerator
            return _inlineCallbacks(None, gen, Deferred())
        --- <exception caught here> ---
          File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1
128, in _inlineCallbacks
            result = g.send(result)
          File "C:\Python27\lib\site-packages\labrad\gpib.py", line 293, in conn
ectToDeviceManager
            manager = self.client[self.deviceManager]
          File "C:\Python27\lib\site-packages\labrad\wrappers.py", line 553, in
__getitem__
            return self.servers[key]
          File "C:\Python27\lib\site-packages\labrad\support.py", line 181, in _
_getitem__
            return dict.__getitem__(self, k)
        exceptions.KeyError: 'GPIB Device Manager'
```
@maffoo?

maffoo, 2016-06-24:
@jwenner, immediately before the line in gpib.py where the error is being thrown, there's a commented-out call to [`self.client.refresh()`](https://github.com/labrad/pylabrad/blob/master/labrad/gpib.py#L292). Uncommenting that would ensure that the client's list of connected servers is up to date, which should prevent the particular exception you are seeing. Can you try uncommenting that line and trying again to see if the autodetection works with different startup orders? I'm not sure this will fix the problem entirely, but it should help.

jwenner, 2016-06-25:
@maffoo, good call! Your suggestion of uncommenting out the line in `gpib.py` fixed the autodetection with all startup orders. I've submitted a pylabrad PR to do this in general.

closed 2016-06-25 by jwenner