labrad/servers #355: ADC LED Questions

amopremcak, 2016-06-30:
Two of our ADCs are displaying some strange behavior. One ADC is build #1, the other is build #7. Before plugging in the 10 MHz reference clock, all 6 LEDs on our ADCs are illuminated. Below are photos:

Build # 7 ADC without reference clock:
[image]
Build #1 ADC without reference clock:
[image]
 
Once we plug the 1.5 V peak-to-peak 10 MHz reference into clock in, the LED at the back r.h.s of the board goes off as well as one of the other LEDs (the group of four) which are related to the PLL I believe. Below are photos:

Build # 7 ADC with reference clock:
[image]

Build # 1 ADC with reference clock:
[image]

Our connections were to the screw terminals were replaced with the parts you recommended:
[image]

Why would plugging in the 10 MHz reference clock signal effect the LEDs on the r.h.s of the board? Have you guys ever seen this type of behavior before?

DanielSank, 2016-06-30:
This isn't super surprising. Check the current limit on your power supply,
probe the voltages on the caps on the power supply section of the board,
and if you find that the voltage is dropping when you turn on the 10 MHz
reference then increase the supply voltage a bit.

<https://cloud.githubusercontent.com/assets/14102485/16506033/8ee77f80-3ee5-11e6-86d6-25ad0e8ad814.JPG>
<https://cloud.githubusercontent.com/assets/14102485/16506084/c0f96d26-3ee5-11e6-82f8-73e32a5ef236.JPG>

<https://cloud.githubusercontent.com/assets/14102485/16506146/351f7f10-3ee6-11e6-9aaa-21c9da99d6be.jpg>

<https://cloud.githubusercontent.com/assets/14102485/16506175/5fc19d8e-3ee6-11e6-85c8-1b1c3d31fe9b.JPG>

<https://cloud.githubusercontent.com/assets/14102485/16506209/841fc00c-3ee6-11e6-8393-aa89ff956a51.JPG>

DanielSank, 2016-06-30:
Whenever anything weird happens with these boards it's a power issue.

amopremcak, 2016-06-30:
Yea I haven't bothered you guys about them for a while. I took your words to heart and fixed up the insides of our power supplies and our cabling from the power supplies to the boards. I am finally confident in our setup but this was a new failure mode. The supply isn't hitting the current limit as it is not beeping/blinking. I will try cranking up the voltage corresponding to this LED once I observe the effect you are talking about. The strange thing is that this isn't happening on either of the two DACs we have on this rack. Thanks for the feedback @DanielSank

amopremcak, 2016-06-30:
Your suggestion did the trick @DanielSank . When hooked up to the 10 MHz reference clock, the pin tied to the 1.5 V supply dropped by roughly 300 mV. Any idea of why that ought to occur?

DanielSank, 2016-07-01:
Higher current draw plus nonzero resistance of cables etc.

amopremcak, 2016-07-01:
Word.

amopremcak, 2016-07-01:
We are having a hard timing getting the LEDs labeled Dt1 A - Dt4 -A on the ADCs to go off after running the DAC bring up script. The number of times we have to run the ghz_dac_bringup.py script seems to vary with the power level at the CLK IN port on the boards. Our cables are equal lengths from the same 10 MHz reference clock. We have varied the power from 1.5 V peak-to-peak to 0.7 V peak-to-peak. What power level do you typically go with?

Also, do you guys use any higher level servers/scripts to interface with your DACs/ADCs above and beyond the ghz_fpga_server? I guess I am just really confused at how to proceed. Robert has us trying to bring up randomized bench marking using these boards but we don't even have full control over the hardware yet. I suspect that some of the custom code that we are using to control these boards is the source of our problems.

amopremcak, 2016-07-27:
Any feedback on the getting the Dt1A-Dt4A lights to go off would greatly appreciated.