labrad/servers #356: Scope binary parsing problem

BrooksCampbell, 2016-07-05:
I finally tracked my scope issue down to bad parsing of the binary waveform data.  There is a function 'query_binary_values" built in to the VISA driver that queries data and parses it correctly.

I've added this function call to the gpib_server.py file and I have successfully tested this fix with the scope accessing this function directly through the gpib_bus:
'''
cxn = labrad.connect()
scope = cxn.collateral_gpib_bus
scope.address(scope.list_devices()[-1])
scope.query_binary_values(':WAVeform:SOURce CHANnel4;DATA?', "h", True)
'''

However, when I went to implement this query in the agilent_infiniium_scope server, it doesn't seem to know about the new query_binary_values function.  Is there some other class definition or something I need to update to add this query_binary_values function to the agilent_infiniium_scope server?

BrooksCampbell, 2016-07-05:
https://github.com/martinisgroup/servers/tree/u/Brooks/issue%23356

BrooksCampbell, 2016-07-05:
gpib_server.py change:
https://github.com/martinisgroup/servers/tree/u/Brooks/issue%23356

BrooksCampbell, 2016-07-05:
I think I found what I need to change in pylabrad @thchwhite helped me out.  I'm going to do that and I'll submit pull requests when I get it working.

BrooksCampbell, 2016-07-06:
gpib_server.py change:
https://github.com/martinisgroup/servers/tree/u/Brooks/issue%23356
https://github.com/labrad/pylabrad/tree/u/Brooks/issue%23356

maffoo, 2016-07-06:
We tracked this down to the fact that the gpib server tries to strip trailing termination characters like `\r` and `\n` (https://github.com/martinisgroup/servers/blob/master/gpib_server.py#L183). This does not work if trying to read binary data, since trailing bytes that just happen to correspond to these termination characters can be dropped. I think the solution to this is to add a separate `read_raw` setting to be used for reading raw bytes, versus the `read` setting that reads string responses. This mirrors the underlying pyvisa interface as well (https://github.com/hgrecco/pyvisa/blob/master/pyvisa/resources/messagebased.py#L288).

Note also labrad/pylabrad#268 which describes a way to hopefully make it easier to fix problems in the gpib system by keeping the bus and device manager servers close to the gpib device server code.