labrad/servers #374: matplotlib ver 1.5.1 doesn't work with the ADR gui

BrooksFoxen, 2016-10-12:
Just documenting an issue a couple of us have seen with the ADR GUI where the embedded plots will not display.

matplotlib version 1.4.0 works fine, 1.5.X doesn't.  I don't know the exactly version number that breaks this.

jwenner, 2016-10-12:
You might check to see if you get a similar warning as with #359.

DanielSank, 2016-10-20:
Are there any error messages?

brooksfoxen, 2016-10-20:
I didn't troubleshoot this very much, because I recalled @andrewdunsworth or @thchwhite running in to this on Quaid's computer and 'fixing' it by downgrading matplotlib.

thchwhite, 2016-10-25:
> Yes, I don't remember what the fix was here...

Downgrading was the fix if I recall.  Its hard to troubleshoot why the
newer version doesn't work because I think its a subtle rendering issue,
where the code thinks its working but nothing gets displayed. Not sure how
to fix this one since Peter coded a lot of this on his own.