labrad/servers #382: Having trouble getting SMA cold switch to work with server.

roboguy222, 2016-10-22:
We are trying to use your server to switch channels on the SMA relays, but the box does not seem to be responding to RS232 signals.  We have checked that the signal actually gets to the box.  Any idea what we should try?  The LED is supposed to light up upon receiving a signal, right?

thchwhite, 2016-10-25:
Hi Chris,

Yes the LED should light up and the box should beep.  If you are confident
that the serial signal is being sent over the wire, there is one other
thing you can try.  We have occasionally seen the error where the box
itself stops responding.  You can fix this by power cycling the
micro-controller.  To do this just disconnect the positive terminal from
the battery then reconnect it.  If the box starts making a constant beeping
noise simply move to any switch channel and flip the set or reset switch.
This will stop the constant beep.

roboguy222, 2016-10-26:
Power cycling did not help (I actually tried that before).  If I run the test script on your wiki and send 'S', it says I don't get any data back.  Let me know if you have any other ideas.

thchwhite, 2016-10-26:
Hmm that is interesting.  My first instinct is that it would be a problem with the serial port/server.  I've seen the com numbers randomly change in windows before.  If you haven't done this it would be good to verify the port by unplugging the usb-serial converter and seeing the port disappear in software.

Do you have more than one switch box?  That would be the easiest way to isolate the switch box as the main problem.  You could simply plug in a different unit and see if it worked.

I also assume the switch box works normally in manual mode?  So you are able to set and reset with the switches on the front.

If both of those are the case, then the only thing I can think of is that maybe the the code is corrupted somehow (which you might be able to fix by re flashing the micro-controller) or the serial chip is broken.  Not sure how likely either is but I haven't seen an issue that wasn't solved by power cycling before.

I'll keep following the bug if you want to run an idea by me or I think of anything new to try.

roboguy222, 2016-10-26:
The usb to serial converter we are using has 8 serial ports.  Others are being used for other purposes and seem to work fine, and I have tried using multiple ports on this one for the cold switch.  I know I am using the right ones because a) the LEDs next to it blink when signals are going out and b) I can see the signal come in on the cold switch box side with an oscope.  I'll try some of our other boxes and see if the problem is with all of them.  Thanks for the help!

thchwhite, 2016-10-30:
No problem,  Hopefully we can get to the bottom of this.

-Ted