labrad/servers #383: Deconvolution - Settling and Reflection rates and amplitudes seem incorrectly applied.

btchiaro, 2016-10-25:
It appears as though settling amplitudes are being applied with the reflection rates.  It's possible that this output is correct and we don't fully understand the physics.  If so please advise.  Our data is below.
Running spectroscopy_z_func on master pyle, master servers, we observe the following pulses, where the magenta trace is the pulse to which the corrections should be applied:

Our control experiment has all amplitudes set to zero, so that there should be no compensation for settling times or reflections.
With parameters:
settlingRates:[0.1 GHz]
settlingAmplitudes:[0]
reflectionRates:[0.001 GHz]
reflectionAmplitudes:[0]
We observe:
[image]

Now we set a large reflection amplitude and do not observe a correction to the pulse.
With parameters:
settlingRates:[0.1 GHz]
settlingAmplitudes:[0]
reflectionRates:[0.001 GHz]
reflectionAmplitudes:[0.5]
We observe:
[image]

If we turn on the settling amplitude, with the reflection amplitude set to zero, we observe what appears to be a reflection correction at a time that seems related to the reflection rate.
With:  
settlingRates:[0.1 GHz]
settlingAmplitudes:[0.5]
reflectionRates:[0.001 GHz]
reflectionAmplitudes:[0.0]
We observe:
[image]

We check this by halving the settling amplitude and see that the correction pulse is halved.
With:
settlingRates:[0.1 GHz]
settlingAmplitudes:[0.25]
reflectionRates:[0.001 GHz]
reflectionAmplitudes:[0.0]
We observe:
[image]

By changing the reflection rate we are able to change the timing of the correction pulse.
With:
settlingRates:[0.1 GHz]
settlingAmplitudes:[0.25]
reflectionRates:[0.002 GHz]
reflectionAmplitudes:[0.0]
We observe:
[image]

@zchen088 @maffoo @JulianSKelly @DanielSank 
Thanks!

zchen088, 2016-10-25:
Ah yes, I found a typo.

zchen088, 2016-10-25:
Note that this server actually lives in pyle, and I've fixed it there.

closed 2016-10-25 by zchen088

btchiaro, 2016-10-25:
Thanks for the fix Jimmy! I'm pulling this now.