labrad/servers #385: why the DAC/ADC board needs so many DC powers?

ypzhong, 2016-11-02:
Hi,

    I'm curious why the DAC/ADC board needs so many different DC powers? I read the document of the voltage regulators, it looks like the input voltage can be much higher than the expected output voltage, for example, we can input 5.4V to all the power pins and the voltage regulators will output 1.2V, 1.8V, 2.5V, 3.2V, 5.0V etc respectively. So why do we use 1.6V, 2.2V, 2.9V, 3.7V, 5.4V DC power supply to power each pin separately? Is there a good reason to do so? Thanks.

Youpeng

ejeffrey, 2016-11-03:
There are basically two reasons for this.

The first is to minimize power dissipation in the regulators.  The idea is
that temperature change is a major cause of drift, so if you have a bunch
of regulators dissipating a few watts each on the PCB, that is potentially
bad.

The second is because some of the power rails are analog supplies and some
are digital.  By using separate supplies for each voltage rail, we keep
those separated.

Check the max power dissipation of the regulators, as long as that is OK
the boards should work with the single supply, but keep in mind that
performance may be changed.

Evan

ypzhong, 2016-11-03:
@ejeffrey  Thanks for your reply. I looked at the document of the regulators, it seems that the regulator heat dissipation is I_OUT*(V_IN-V_OUT)+I_GND*I_IN, so the larger the input/output voltage differs the more heat the chip dissipates.

closed 2016-12-02 by ypzhong