labrad/servers #386: Can't get trace from scope.

andrewdunsworth, 2016-11-17:
[image]

maffoo, 2016-11-17:
What version of pylabrad are you using on the computer where the server is running? The `read_raw` method was added in 0.96.1 (https://github.com/labrad/pylabrad/commit/b9067425ce1bddf9efa08638e89b0395e52fc52b).

andrewdunsworth, 2016-11-17:
I just pulled master.

andrewdunsworth, 2016-11-17:
On gordy.

maffoo, 2016-11-17:
From the traceback, it looks like python was installed with pip (`C:\Python27\lib\site-packages\labrad\server.py`). You can check the version by calling `pip list` in a terminal windows and looking for the pylabrad entry. You can upgrade it with `pip install --upgrade pylabrad`.

DanielSank, 2016-11-17:
I think Matthew means that pylabrad was installed with pip.

maffoo, 2016-11-17:
Yes, thanks @DanielSank :-)

DanielSank, 2016-11-18:
So... this went away?

maffoo, 2016-11-18:
Yes, updating pylabrad fixed things. (This also required some futzing with the python path because windows.)

closed 2016-11-18 by maffoo