labrad/servers #389: what does "bad clocking phase" mean?

ypzhong, 2016-12-02:
Hi,

    I found that the LED[1] of our ADC board is always on, and I look up the document, it says it indicates "bad clocking phase of either AD chip", can anyone explain a little bit what does it mean, and how to fix it? Thanks.

Best,
Youpeng

ejeffrey, 2016-12-05:
I am not sure what this means, I have never looked at those LEDs.  Possibly
@DanielSank knows

However, there is one issue that I know about that may be the problem.  The
ADC outputs  data at 500 MHz on a 16 bit data bus containing two samples
(to reach the total 1 GS/s rate).  There is an ambiguity here, relative to
the FPGA, the ADC can be generating samples on "even" or "odd"
nanoseconds.  The FPGA only latches on "even" nanoseconds, so if the ADC
clock phase is wrong, you will get data corruption errors.  We don't have
any direct control over this, it simply depends on the state of an internal
clock divider inside the ADC.

You can confirm that this is the case by recording a time trace (aka
"average" mode) with averages set to '1'.  If one ADC is in the "bad"
state, you will see very obvious bit flip errors when you plot I/Q vs. time
-- especially if the signal is close to zero.

The only way to fix this currently is to run the PLL init repeatedly, which
will cause the ADC clock divider to skip phases.  You just have to do this
over and over until both ADCs have the right phase.

You are supposed to be able to get this information from the register
readback -- I don't know if that is the same information as the LED you are
seeing.

Evan

DanielSank, 2016-12-05:
Off the top of my head, I do not know what those LED's mean. This would have to be checked with the FPGA program and circuit diagram. I'll see if we can figure it out from our documents.