labrad/servers #391: master DAC board timeout while slave ADC board doesn't

ypzhong, 2016-12-08:
Hi,
    Recently I'm trying to use the labRAD server downloaded from github, and I have some problem with the DAC/ADC boards. When I use one DAC board to trigger the ADC board and run the board group (which only has one DAC and one ADC), I get timeout error very frequently, and it's the DAC (the master board) that has timeout (the ADC always works). Changing to another DAC board is the same. However I cannot locate the position where the timeout error comes from so I don't know what's wrong. @maffoo @ejeffrey @DanielSank Do you have some idea under what circumstance the master board has timeout while the slave board is normal? I really appreciate your help. Thanks.

maffoo, 2016-12-09:
If you're using DACs and ADCs together all the data is coming back from the ADC boards. Because there's nothing being returned from the DAC boards they can't really timeout (it takes no time to collect 0 packets). What might be happening is that the timeout error message is misleading; for example, there is a known issue with reading back the execution counter on the DACs because an attempt to read the execution counter actually resets the counter so it always reads as 0, even if the board executed the sequence properly.

The timeout means that the ethernet server did not collect the expected number of packets from the ADC within the expected time. So either the packets did not get sent, or they got dropped on the wire and were not received. Packets may not get sent if the board didn't execute for some reason, for example due to a daisy chain connection problem, so you'll want to check the daisy chain connection. Also, it can be very helpful to use a program like wireshark to capture the ethernet packets and inspect them to see if they were actually sent by the ADC. There are lots of other things one could try, but hopefully that can help you get started troubleshooting.

ypzhong, 2017-01-11:
Thanks @maffoo , I struggled for a while but didn't figure out how to solve this problem. Recently we used a separate local network for the custom boards and the timeout issue disappeared surprisingly. Anyway I don't need to worry about it anymore.

closed 2017-01-11 by ypzhong