labrad/servers #397: 1 GHz DAC serializer PLL Question

amopremcak, 2017-01-06:
I am seeing an issue on one of our custom electronics racks that is forcing me to run the ghz_dac_bringup.py script after every measurment. When I bring up the boards, the LEDs Dt4 A through Dt1 A all go off and the LEDs indicating good power are illuminated. At the end of a measurement script, all DACs will usually have LED Dt4 A illuminated which, according to the documentation, means that we are not locked to the internal 1 GHz DAC serializer PLL. In these measurements, we are not using the ADCs. The 10 MHz reference level is ~8 dBm at the CLK IN port. The input power levels were probed and are greater than the recommended levels by a few hundred mV just to be safe. Has anyone encountered this before? Do my clock levels seem too high?

amopremcak, 2017-01-06:
Also, this LED becomes illuminated in the absence of any board communications. That is, I can just run bring up, wait a minute or so and it turns back on.

ypzhong, 2017-01-11:
I think the 10MHz clock signal level is OK, did you mean all the DAC boards on the same rack have the same problem or only one DAC board that has this problem? If all the DAC boards have the same problem, I doubt it's the clock source problem.

amopremcak, 2017-01-11:
I'm still working on diagnosing the issue. There are times when all LEDs illuminate and other times when only a subset do. Changing the boards device IDs seemed to have an effect, namely the time it takes for these LEDs to latch is quite a bit longer and is sometimes stable overnight. I have found that when the LED latches, it leads to some pulse distortion and timing slips between boards.