labrad/servers #409: ADR GUI sometimes fails to start: "No module named qt4reactor"

mgiustina, 2017-02-03:
Starting the ADR GUI (in `/servers/GUIs/ADR/ADRcontrol2.py`) sometimes evokes the error:
`Traceback (most recent call last):`
`  File "ADRcontrol2.py", line 574, in <module>`
`    import qt4reactor`
`ImportError: No module named qt4reactor`

I haven't figured out how to reliably reproduce the error. Perhaps the error is more likely when the ADR GUI is already running on another machine? But that sometimes also works successfully.

Current fix is to restart all the ADR servers, in the worst case to restart the labradhost machine.

pklimov, 2017-02-03:
looks like you need to install the qt4reactor module. Maybe try:
`pip install qt4reactor`

maffoo, 2017-02-08:
I'm worried about the "sometimes" here. Modules being on the pythonpath should not be something that changes on its own, though I guess this could actually be a problem with runtime initialization of the qt4reactor module, not just finding the code. Are you always starting the GUI program in the same way? Are you using a virtualenv? I don't think restarting servers should have any effect on this, but I guess we need to figure out what the actual problem is.

mgiustina, 2017-02-09:
In that particular case, it was my fault: I wasn't in my virtual env. But it reminded me that both Brooks and I have seen such a problem before and hadn't filed an issue. I'll try to update with as much "more info" as I have when it comes up again.