labrad/servers #410: Current data_vault server version

zibrov-zlobin, 2017-02-06:
Hi,
I was wondering what the current version of the data_vault server (whats the difference from multihead version?)

I am trying to setup a fresh installation of labrad:
scalalabrad  0.8.0
pylabrad 0.96.4 

and the current version of datavault from: martinisgroup/servers/

the datavault freezes up after the entering a login password. Happens for both a global user and a newly created user.
anything else i should setup before using? 
thanks

maffoo, 2017-02-08:
The data vault server ([data_vault.py](https://github.com/martinisgroup/servers/blob/master/data_vault.py)) and the multi-headed server ([data_vault_multihead.py](https://github.com/martinisgroup/servers/blob/master/data_vault_multihead.py)) use the same backend code for actually storing data on disk. The difference is that the multiheaded version can connect to multiple labrad managers, so it's only needed if you have a setup where you are running multiple managers and want them all to have access to the same datasets.

As for setup, when you first run the data vault it should ask you to enter a path where the data should be stored and it saves this path in the registry. I don't know exactly why it would be hanging; one thing to check would be whether you are able to connect to the labrad manager directly from python with `labrad.connect()`.

zibrov-zlobin, 2017-02-14:
yeah, the connection to labrad works directly from python and all other servers. 
the new data vault doesnt offer to create a repository. Taking an older version of data_vault worked - created a new repository and connected fine.

closed 2017-04-24 by zibrov-zlobin