labrad/servers #419: Unable to connect to datavault - No section 'File System'

btchiaro, 2018-01-06:
I am currently unable to connect to the datavault either through an ipython session or the grapher.  It looks like a file system issue.  Please advise.

Unable to connect to Data Vault
org.labrad.errors.LabradException: 0: [Data Vault] Remote Traceback (most recent call last): File "/usr/local/lib/python2.7/dist-packages/labrad/server.py", line 262, in handleRequest result = yield setting.handleRequest(self, c.data, data) File "/usr/local/lib/python2.7/dist-packages/labrad/decorators.py", line 194, in handleRequest return f(self, c, *data) File "/usr/local/src/servers/datavault/server.py", line 121, in cd _session = self.session_store.get(temp) # touch the session File "/usr/local/src/servers/datavault/__init__.py", line 111, in get session = Session(self.datadir, path, self.hub, self) File "/usr/local/src/servers/datavault/__init__.py", line 140, in __init__ self.load() File "/usr/local/src/servers/datavault/__init__.py", line 156, in load self.counter = S.getint(sec, 'Counter') File "/usr/lib/python2.7/ConfigParser.py", line 359, in getint return self._get(section, int, option) File "/usr/lib/python2.7/ConfigParser.py", line 356, in _get return conv(self.get(section, option)) File "/usr/lib/python2.7/ConfigParser.py", line 607, in get raise NoSectionError(section) ConfigParser.NoSectionError: No section: 'File System'

btchiaro, 2018-01-06:
This might be due to the datavault being out of memory.   Trying to open a dataset on a different computer in the delphi grapher gives this error.
[image]

amitsv1, 2018-01-07:
As the grapher said, there was no space left on the device because it got filled up.

closed 2018-01-07 by amitsv1