labrad/servers #422: GPIB server issue with PNA_X server

thchwhite, 2018-06-20:
`
[PNA_X] Remote Traceback (most recent call last):
Failure: labrad.types.types.Error: (0) [pfeiffer-lab GPIB Bus] Remote Traceback (most recent call last):
  File "/usr/local/labrad/pyle/labrad/server.py", line 255, in request_handler
    c.data, flat_data)
  File "/usr/local/labrad/pyle/labrad/server.py", line 216, in _dispatch
    return func(*args, **kw)
  File "/usr/local/labrad/pyle/labrad/decorators.py", line 200, in handleRequest
    return self.func(server, c, data)
  File "gpib_server.py", line 293, in read
    return ans.decode('utf-8').strip()
  File "/usr/lib64/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
exceptions.UnicodeDecodeError: 'utf8' codec can't decode byte 0xbf in position 7: invalid start byte
 [payload=None]
 [payload=None]
'

maffoo, 2018-06-20:
We're probably doing a read that should be read_raw.

thchwhite, 2018-06-20:
At some point I believe this got fixed in the GPIB server.  Its possible the commit was never pushed though.

kunalq, 2018-06-21:
Closing, as this is related to another repo. :)

closed 2018-06-21 by kunalq